joan2937/pigpio #350: Unexpected callback behavior

BTPankow, 2020-05-22:
The following code seems to trigger a callback, when I would expect no callback to be triggered.

I've been able to reliably cause this when running the script from a remote computer connected to the pi, but have also seen this on the pi itself (but much less often). It seems like it might have to do with setting the glitch filter since I have not been able to reproduce the behavior with the glitch filter lines removed.

I haven't done a full investigation yet, but I would like to get some feedback on whether this is expected or not.

I'm using a raspberry pi model 3B+ with pigpio version 1.46 and python 2.7.

This is what I see when I run remotely (I would expect to see no output):
```
$ python test.py
gpio = 25, level = 0, tick = 2990614791
gpio = 25, level = 1, tick = 2990638556
```

```python
import pigpio
import time

def cbf(gpio, level, tick):
    print("gpio = {}, level = {}, tick = {}".format(gpio, level, tick))

if __name__ == "__main__":
    pi = pigpio.pi("pi")

    pi.set_mode(24, pigpio.INPUT)
    pi.set_pull_up_down(24, pigpio.PUD_OFF)
    pi.set_glitch_filter(24, 0)
    pi.set_mode(24, pigpio.INPUT)
    pi.set_pull_up_down(24, pigpio.PUD_UP)
    pi.set_glitch_filter(24, .1*1e6)

    pi.callback(24, pigpio.EITHER_EDGE, cbf)

    pi.set_mode(25, pigpio.INPUT)
    pi.set_pull_up_down(25, pigpio.PUD_OFF)
    pi.set_glitch_filter(25, 0)
    pi.set_mode(25, pigpio.INPUT)
    pi.set_pull_up_down(25, pigpio.PUD_UP)
    pi.set_glitch_filter(25, .1*1e6)

    pi.callback(25, pigpio.EITHER_EDGE, cbf)

    # Phantom callback usually occurs within the first 2s
    time.sleep(2)
```
(This code is a distillation of what [gpiozero](https://github.com/gpiozero/gpiozero) is doing when setting up a Button with a pigpio backend, so it might look a little strange)

Thanks for this great library, I appreciate the great documentation and hardware capabilities.

guymcswain, 2020-05-22:
The first time the glitch filter is set, the input is floating and the steady argument is zero, meaning no filter.  I'm not sure what the first filter was meant to accomplish but it seems to be the source of your concern.

guymcswain, 2020-05-22:
Also, it might help to understand that once you tell pigpio to begin processing pin changes (callback, alerts, notifications) it is looking back in time by 100 msec or so.

BTPankow, 2020-05-23:
Okay, thanks for the info, that is helpful to know about the 100ms look back in time.

What I don't get is that I don't see this when the callback is set without the glitch filter this doesn't happen, but the glitch filter does cause this to happen. Is this because the glitch filter reads the 'old' state when it is initialized, causing it to trigger the callbacks once the state almost immediately changes?
(https://github.com/joan2937/pigpio/blob/master/pigpio.c#L12338)

As far as your first comment, the code was distilled from what gpiozero does when setting up a button, so that's why it may look a little strange with that floating GPIO and 0 steady glitch filter (but like you said, that should be okay since it basically ignores it). I agree that another solution for debouncing would likely be better when using pigpio for gpiozero.

Thank you for the help, I appreciate it.

BTPankow, 2020-05-23:
I did a little more investigation and found that I could also trigger the callback with the following code. Not as often, but still more than I would expect.
```
#include "pigpiod_if2.h"
#include <stdio.h>
#include <unistd.h>

void cbf(int gpio, int level, unsigned int tick) {
    printf("gpio = %d, level = %d, tick = %u\n", gpio, level, tick);
}

int main() {
    gpioCfgInternals(984762879, 4);
    gpioInitialise();
    gpioSetMode(24, PI_INPUT);
    gpioSetPullUpDown(24, PI_PUD_UP);
    gpioGlitchFilter(24, 100000);
    gpioSetAlertFunc(24, cbf);
    sleep(1);
    gpioTerminate();
}
```

There are a few things in the glitch filter code that I noted.

When checking if the time between a change in value and the steadyUs, there's a comparison between an `int` and an unsigned int (`uint32_t`). https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5719
This means that if the `diff` (time between last sample and last change) is negative, it will appear larger than the `steadyUs`, which seems like it might not be desired. Normally, I wouldn't expect that to happen since the samples should be processed in increasing tick values (if I understand correctly), but this can happen when the glitch filter is first initialized for a GPIO.

The filter will set the last changed time using the system clock (`systReg[SYST_CLO]`), which may be farther ahead than the samples processed (and usually is). This means that the `diff` is usually negative when first calculated, which makes it appear larger than `steadyUs`, which causes the current bit value to marked as stable.

However, on rare occasions, this clock is not always ahead of the next sample to be processed, which causes the glitch filter to set the sample level to the opposite of what it currently is (since the `diff` is seen as less than `steadyUs`).

One solution would be to not initialize the glitch filter states in `gpioGlitchFilter` and instead when we first process the samples in `alertGlitchFilter` (using the first sample values). This way we ensure that the clock lines up with the start of the processed samples.

I've attached two logs, one that exhibits the behavior I'd want, with no glitch filter triggering a callback (`good.log`) and one with the glitch filter triggering a callback (`bad.log`). The logs were created with added `DBG` statements in the `pigpio.c`.

[bad.log](https://github.com/joan2937/pigpio/files/4672095/bad.log)

[good.log](https://github.com/joan2937/pigpio/files/4672096/good.log)

BTPankow, 2020-05-24:
Just a little more info, this is one solution to the issue. With this patch, I have been unable to reproduce the issue.

I wanted to avoid changing too much here, so I didn't add anything for checking if the `diff` is negative before comparing to the `steadyUs`. If the `diff` is could somehow be negative, that could still be a problem. Also, when initializing the glitch filter using the first sample, the first sample is still processed later on; that could be changed too.

```patch
diff --git a/pigpio.c b/pigpio.c
index 1bdae12..c825696 100644
--- a/pigpio.c
+++ b/pigpio.c
@@ -950,6 +950,7 @@ typedef struct
    uint32_t nfRBitV;
 
    uint32_t gfSteadyUs;
+   uint8_t  gfInitialised;
    uint32_t gfTick;
    uint32_t gfLBitV;
    uint32_t gfRBitV;
@@ -5683,7 +5684,7 @@ unsigned alert_delays[]=
 static void alertGlitchFilter(gpioSample_t *sample, int numSamples)
 {
    int i, j, diff;
-   uint32_t steadyUs, changedTick, RBitV, LBitV;
+   uint32_t steadyUs, changedTick, RBitV, LBitV, initialised;
    uint32_t bit, bitV;
 
    for (i=0; i<=PI_MAX_USER_GPIO; i++)
@@ -5696,6 +5697,17 @@ static void alertGlitchFilter(gpioSample_t *sample, int numSamples)
          RBitV       = gpioAlert[i].gfRBitV;
          LBitV       = gpioAlert[i].gfLBitV;
          changedTick = gpioAlert[i].gfTick;
+         initialised = gpioAlert[i].gfInitialised;
+
+         if (!initialised && numSamples > 0)
+         {
+            /* initialise filter with first sample */
+            bitV = sample[0].level & bit;
+            RBitV = bitV;
+            LBitV = bitV;
+            changedTick = sample[0].tick;
+            gpioAlert[i].gfInitialised = 1;
+         }
 
          for (j=0; j<numSamples; j++)
          {
@@ -12333,18 +12345,8 @@ int gpioGlitchFilter(unsigned gpio, unsigned steady)
 
    if (steady)
    {
-      gpioAlert[gpio].gfTick  = systReg[SYST_CLO];
-
-      if (gpioRead_Bits_0_31() & (1<<gpio))
-      {
-         gpioAlert[gpio].gfLBitV = (1<<gpio);
-         gpioAlert[gpio].gfRBitV = 0 ;
-      }
-      else
-      {
-         gpioAlert[gpio].gfLBitV = 0 ;
-         gpioAlert[gpio].gfRBitV = (1<<gpio);
-      }
+      /* initialise values next time we process alerts */
+      gpioAlert[gpio].gfInitialised = 0;
    }
 
    gpioAlert[gpio].gfSteadyUs = steady;
```

guymcswain, 2020-05-24:
I'm lost.  Your original question was whether the behavior is expected and my response was to let you know that is.  Are you now saying there is a problem in the library?   If so please state the problem?

BTPankow, 2020-05-24:
I would recommend checking the two logs I attached to see the issue in action, but basically the glitch filter gets initialized with a tick count that is usually greater than the tick count when the samples are processed for alerts.
This causes the time difference to usually be negative, but sometimes positive when checked in the `alertGlitchFilter` function after first initialization. The steady state checking in the glitch filter act in a slightly different way with a negative vs. positive time difference.

Sorry if it's still not clear, just let me know what areas of my comments I can go into more detail on.

BTPankow, 2020-05-24:
But, yes, I am saying there is a problem in the library that sometimes causes the level to be reported incorrectly for alerts (like callbacks) when a glitch filter for a pin is created.

BTPankow, 2020-05-24:
Just to try to add a little more info, here is what usually happens:

1. Glitch filter gets initialized with a tick value here: https://github.com/joan2937/pigpio/blob/master/pigpio.c#L12336
2. A diff between that initial tick value and the sample tick value is created here:
https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5717
3. Since the initial tick value is usually greater than the sample tick value, the diff is negative.
4. This means that this statement will evaluate to true (negative diff > steady time):
https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5719
5. This marks the sample level as stable here:
https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5722
6. No alert for callbacks is created since the sample level is not flipped.

However, sometimes the initial tick value is smaller than the sample tick value, causing the diff to be positive, meaning that the sample level is flipped (incorrectly) here:
https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5728
This fires a callback incorrectly.

guymcswain, 2020-05-25:
You need to demonstrate "the problem" with a test case.

BTPankow, 2020-05-25:
I might be misunderstanding what you were commenting on for the "original behavior". Should just setting a glitch filter on a gpio pin trigger a callback?

This script does trigger callbacks for me even though the pin should have been in a pulled up input state "a while ago". I would imagine the latency you mentioned before shouldn't be a factor here?

```python
import pigpio
import time

def cbf(gpio, level, tick):
    print("gpio = {}, level = {}, tick = {}".format(gpio, level, tick))

if __name__ == "__main__":
    pi = pigpio.pi("pi")

    pi.set_mode(24, pigpio.INPUT)
    pi.set_mode(25, pigpio.INPUT)
    pi.set_pull_up_down(24, pigpio.PUD_UP)
    pi.set_pull_up_down(24, pigpio.PUD_UP)

    # Wait long enough for gpio to get in a steady state
    time.sleep(10)

    pi.set_glitch_filter(24, .1*1e6)
    pi.callback(24, pigpio.EITHER_EDGE, cbf)

    pi.set_glitch_filter(25, .1*1e6)
    pi.callback(25, pigpio.EITHER_EDGE, cbf)

    # Phantom callback usually occurs within the first 2s
    time.sleep(2)
```
```
$ python glitch_test.py
gpio = 25, level = 0, tick = 1633872941
gpio = 25, level = 1, tick = 1633880090
```

guymcswain, 2020-05-25:
Thanks.  Yes I would consider this time of scenario to be an issue.  It looks like you missed the pull-up on pin 25.  Can you reproduce with that fixed?

BTPankow, 2020-05-25:
Good catch on the pull up, I fixed that and was still able to reproduce the issue on my side.

guymcswain, 2020-05-25:
Do you have any other up activity going on?  Those pull-ups are weak and still susceptible to internal and external noises sources.  You can try using a 5k to 10k resistor to be sure the problem is not noise.

BTPankow, 2020-05-25:
Nothing else that I know of, but I can try with external pull ups and let you know. Have you tried to run the example script on your side?

BTPankow, 2020-05-25:
Okay, I just tested with pull down and pull up (10k) resistors through a breadboard to 3v3 and ground and have been able to reproduce the issue.

guymcswain, 2020-05-25:
Not yet.  I'll do so later today.

BTPankow, 2020-05-25:
Okay, thanks, it's strange. I also just tested with a pi zero I had around with the latest pigpiod and was able to see the same behavior. (I was running with `sudo pigpiod -g` if that does make a difference)

guymcswain, 2020-05-25:
Until I can run your script locally, please run the x_pigpio.py test.  It also uses gpio 25 with nothing connected.

BTPankow, 2020-05-25:
I ran the test locally on the pi as well as remotely and had the tests pass.

I ran into this failure the first time I ran locally, but then I ran the tests a couple more times and had them all pass each time.
`TEST 13.12 FAILED got 1 (noise filter, wave chain, tally: 0)`

guymcswain, 2020-05-25:
I ran your script 10 times without any callbacks occurring.  Unrelated, I just changed this line:
` pi = pigpio.pi("pi")` to this:
`pi = pigpio.pi()`

BTPankow, 2020-05-25:
Thanks for checking, I was seeing this behavior when running on a remote computer which might be the difference.

BTPankow, 2020-05-25:
I did see it locally too, but it was much, much less frequent. Also, this is likely unrelated, but I was running pigpio with `sudo ./pigpiod -g` (not through systemctl or in the background).

guymcswain, 2020-05-25:
Ok, I see it running remotely.

guymcswain, 2020-05-25:
I think it may be caused by the `steady` parameter on `set_glitch_filter`.  I'm not reproducing with small values 1, 10.

BTPankow, 2020-05-25:
Right, that make sense because then the `diff` (https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5717)  is likely greater than `steadyUs` on the first glitch filter initialization, causing the sample level to not get flipped. Some of my previous comments touched on this and the cause and effects of the `diff` being positive vs. negative when the `alertGlitchFilter` is run after `gpioGlitchFilter`.

BTPankow, 2020-05-25:
If you apply my patch, the issue should go away. However, I'm still not sure why the tick value here (https://github.com/joan2937/pigpio/blob/master/pigpio.c#L12336) sometimes is less than the first samples processed in `alertGlitchFilter`. Personally, I'd prefer to remove the possibility for timing problems and do the initialization in `alertGlitchFilter`, but it would be nice to know the timing difference.

If somehow `alertGlitchFilter` has a pause in it's execution, that would cause it, but the code seems like it shouldn't have a line that would take a while from what I can tell.

BTPankow, 2020-05-25:
Also, I'm not sure what you mean by the filter window being larger than the script execution time, can you help clarify? The script should wait 2s which should be plenty of time compared to a 20000us `steady` time.

guymcswain, 2020-05-25:
Yeah, just realized those are microseconds.

guymcswain, 2020-05-25:
`alertGlitchFilter` is postprocessing "older" samples while `gpioGlitchFilter` is real-time, maybe?  I'll need some time to look at the source code but don't have plans to do that for the next few days.

I'm holding this as a valid issue.

BTPankow, 2020-05-25:
Yep, that's usually what happens, the initialized tick is greater than the "older" samples causing `diff` to be negative and greater than `steadyUs` (resulting in okay behavior). It's only when the samples are newer in `alertGlitchFilter` that the problem occurs (`diff` less than `steadyUs`).

I think you're on the right track to getting close to my understanding of the problem. Not sure about the underlying timing inconsistencies, so if you help with that, that would be appreciated.

I'll update the title and probably submit a pull request with the current fix and a test to reproduce the issue a little later.

guymcswain, 2020-06-25:
@BTPankow , sorry this has taken some time - I'm just now preparing for the next release (v77).  I reviewed your PR and it appears to be very well done - following the contrib guideline and providing test case(s).  Its not often this happens so a big thank you!

> ...it would be nice to know the cause of the timing difference.

> I'd prefer to remove the possibility for timing problems and do the initialization in alertGlitchFilter ...

I'm satisfied the latter explains the former.

guymcswain, 2020-06-26:
@BTPankow ,
When running `x_pigpio.py` the first time, everything passes.   However, on a second run I'm getting:
```
Traceback (most recent call last):
  File "./x_pigpio.py", line 1059, in <module>
    if '3' in tests: t3()
  File "./x_pigpio.py", line 211, in t3
    CHECK(3, t, int((1E3*(on+off))/on), int(2E7/x), 1, "set servo pulsewidth")
ZeroDivisionError: float division by zero
```
I'm running your branch in isolation - previously I saw this on a branch with all changes, including yours, staged for the next release.  Can you reproduce this?

BTPankow, 2020-06-27:
@guymcswain Thanks for the feedback. I just checked out the tests and you're right, running subsequent tests causes failures.

I had forgotten to clear the glitch filter on the two GPIO pins I was testing with, causing the filter to affect other tests. I just added a commit to fix this issue (https://github.com/joan2937/pigpio/pull/353/commits/26c9ca2bce68686b4ac07d96219658bad61f079a).

I've run the `x_pigpio.py` tests multiple times in a row and have had them all pass each time with this change.

PR #365 (2020-06-28): V77 release
Version 77 is a patch release for bugs and documentation clarifications.

Bug fixes:

- 5395bbe  fix #354 and fix #363.
- 328e09a  fix #350

Documentation:

- 1a98364  fix #355
- 1c5b27f  fix #356 
- d7379ba fix #341

Reverted commits:  (previously on develop branch)

- f51642a  "Remove excessive PROT_EXEC"
- a77351a  "Calling exit from a signal handler is not safe."
closed 2020-06-28 by guymcswain