joan2937/pigpio #351: Trouble updating pigpio on raspberry pi, seeking to use create_wave_and_pad

whogben, 2020-05-22:
I am using pigpio via python on a raspberry pi 4. Initially I used the pre-installed version, but it reports as version 71, and I need create_wave_and_pad() function that does not appear to be available in that version.

I tried using pip and pip3 install pigpio --upgrade, and it appeared to work, but still pigpio reports its version as 71 in python, and it doesn't have create_wave_and_pad

Is there another way to get the latest pigpio, or simply to get access to the functionality of create_wave_and_pad?

(the issue I need create_wave_and_pad for is because otherwise, I cannot create a rolling wave buffer, as the wave_delete function won't work while another wave with a higher ID is playing)

guymcswain, 2020-05-22:
You will need to install the latest version - see this link http://abyz.me.uk/rpi/pigpio/download.html

whogben, 2020-05-22:
Thank you, that seems to have solved the version issue - it now says version 76 instead of 71. However, it didn't solve the missing wave_create_and_pad() function like I expected - this is the output I get:

```
pigpioWaveBuffer: pigpio daemon started
version:76
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/home/pi/Desktop/Body/Parts/Drivers/pigpioStepper/pigpioWaveBuffer.py", line 78, in run
    next_wave = self.pi.wave_create_and_pad(50)
AttributeError: 'pi' object has no attribute 'wave_create_and_pad'
```

To confirm, when I replace "wave_create_and_pad(50)" with "wave_create()" it does work, until it runs out of wave IDs / resources at least. Do you have any idea why it would report version 76, but not be able to find the function?  Thank you for the help so far!

guymcswain, 2020-05-22:
My testing was done with python 2.7.  Is it possible for you to try 2.7?

whogben, 2020-05-22:
Hmm, just tried running it via python 2.7 but got the same issue - the version outputs as 76 but I'm guessing that's the underlying daemon version? The daemon might have updated without the python updating perhaps

```
pi@raspberrypi4-1:~/Desktop/Body/Parts/Drivers/pigpioStepper $ python pigpioWaveBuffer.py 
pigpiod: no process found
pigpioWaveBuffer: pigpio daemon started
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "pigpioWaveBuffer.py", line 78, in run
    next_wave = self.pi.wave_create_and_pad(50)
AttributeError: pi instance has no attribute 'wave_create_and_pad'

```

joan2937, 2020-05-22:
Check your Python pigpio module version.
```
ceres:~ $ python
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pigpio
>>> print(pigpio.VERSION)
1.46
>>> help(pi.wave_create_and_pad)

>>> exit()
ceres:~ $ 
```

guymcswain, 2020-05-22:
@joan2937 ,
Unfortunately, I didn't bump the python module revision in V76.  A return of 1.46 may be a false positive.

guymcswain, 2020-05-22:
@whogben , when you install the latest from the GitHub repo you'll have V76 which includes both the daemon and the python module.  You are running on the same host, yes?

guymcswain, 2020-05-22:
You might also try this:

> To check the library
These tests make extensive use of GPIO 25 (pin 22).  Make sure nothing, or only a LED, is connected to the GPIO before running the tests.  Most tests are statistical in nature and so may on occasion fail.  Repeated failures on the same test or many failures in a group of tests indicate a problem.
```bash
sudo ./x_pigpio # check C I/F

sudo pigpiod    # start daemon

./x_pigpio.py   # check Python I/F to daemon
./x_pigs        # check pigs   I/F to daemon

```

joan2937, 2020-05-22:
@guymcswain 

I take your point.  Buts as long as the system is built form source it should pull in the correct Python module.  A pip install of the pigpio Python module is likely to be several versions behind as it is maintained separately by Ben Nuttall (on behalf of gpiozero).

whogben, 2020-05-22:
Thanks for all the test ideas - results below:

> Check your Python pigpio module version.

```
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pigpio
>>> print (pigpio.VERSION)
1.46
```

> You are running on the same host, yes?
Same device, it's running locally

> You might also try this: To check the library These tests make extensive use of GPIO 25 (pin 22).

```
pi@raspberrypi4-1:~/pigpio-master $ sudo ./x_pigpio
2020-05-22 17:54:07 initInitialise: Can't lock /var/run/pigpio.pid
pigpio initialisation failed.
```

I think it's plausible that in my attempts to install pigpio via pip/pip3 and apt-get I might have gotten the system into an unusual state, where it's finding the right pigpio daemon (hence the v176 reporting) but not finding the right python module for it. I apologize as I'm quite new to this, is there a better way to clear off the system and try again than starting from a fresh raspian install? If that's the recommended next step, I'll try it next. Note that my raspian came with pigpio 71 installed already as a default.

guymcswain, 2020-05-22:
Just run the last two tests since you have the daemon running already.

whogben, 2020-05-22:
The last two tests look good - specifically from ./x_pigpio.py:

```
pi@raspberrypi4-1:~/pigpio-master $ ./x_pigpio.py
Connected to pigpio daemon.

Testing pigpio Python module 1.46
Python 2.7.16 (default, Oct 10 2019, 22:02:15)  [GCC 8.3.0]
pigpio version 76.
Hardware revision 12595474.

...

TEST  5.34 PASS (wave create and pad, wid==: 0)
TEST  5.35 PASS (wave create and pad, wid==: 1)
TEST  5.36 PASS (delete wid==0 success: 0)
TEST  5.37 PASS (No more CBs using wave create: -67)
TEST  5.38 PASS (wave create pad, count==3, wid==: 0)
```

The only failing test was in ./x_pigs:

```
WVCAP-c ok
ERROR: No more CBs for waveform
```

guymcswain, 2020-05-22:
I'll need to check that last error using pigs - may be an oversight, but the first test shows you have the correct python module installed.

whogben, 2020-05-23:
Do you think the WVCAP-c error and the "'pi' object has no attribute 'wave_create_and_pad'" errors are related?

Given it's the right python, and the other tests are passing including the other wave_create_and_pad ones, would you recommend I try a fresh install of the whole system, or another approach?

guymcswain, 2020-05-24:
> would you recommend I try a fresh install of the whole system, or another approach?

If I was in your situation I would probably start over with a fresh install.  Please us know if you continue to have an issue accessing the new API.

closed 2020-06-28 by guymcswain