joan2937/pigpio #352: Changing default SDA and SCL pins for configuring RPI as a i2c slave

tony-teknique, 2020-05-22:
Hi, all:
I am hoping someone will be able to point me in the right direction here. 
I have been playing with the i2c slave example (as documented in the `bsc_i2c` function in `pigpio.py`) and it has been working for me. Now, I would like to change the default SDA and SCL GPIO pins to some other GPIOs (7 and 8) because 18 and 19 have already been taken for some other functions in my project. 

I have tried changing the constant under pigpio.h, `BSC_SDA_MOSI` and `BSC_SCL_SCLK` but it does not seem to change anything (I did recompiled and reinstall).

So I am wondering whether that is the only place I need to change or is there some other lines I have to change to make this customisation work.

guymcswain, 2020-05-22:
The bsc apis access the hardware peripheral.  I think these pins are fixed but if not you would find that information in the bcm2835 specification.  You can find the spec on the raspberrypi.org site under hardware.

tony-teknique, 2020-05-22:
You are right. As Joan pointed out here. https://raspberrypi.stackexchange.com/a/95324
It's fixed on the Hardware so we can't change it

AlessioR, 2020-05-22:
Hi,

sorry to ask here, but I'm trying to achieve same result:
to set as slave a RPi Model B Plus Rev 1.2 (BCM2835), the example in pigpio.py reports:

> This example assumes GPIO 2/3 are connected to GPIO 18/19 (GPIO 10/11 on the BCM2711)

This is intented to do on a single rpi ?

joan2937, 2020-05-22:
@AlessioR 

The example is just an example of usage.  For the example it is simplest to use the same Pi to act as master and slave.

In practice one Pi will act as the I2C slave and another Pi (or another microprocessor) will act as the I2C master.

AlessioR, 2020-05-22:
Thank you.
I was afraid to burn my rpi connecting gpio pins together,
but I've no luck in running the example on samePI.
The command i2cdetect -y 1 reports:
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- 27 28 29 2a -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 70: -- -- -- -- -- -- -- --
>
Why address 0x13 is not showed ?

I've this line in my /boot/config.txt
> dtparam=i2c_arm=on,i2c_arm_baudrate=400000

Thank you.

joan2937, 2020-05-22:
@AlessioR 

I have no idea.  This does not appear to be an issue with pigpio though.

guymcswain, 2020-05-22:
Use at least 1K ohm resistors in series with the wires and you won't 'burn' your gpio.  I found that experimentation is the best learning tool when dealing with the I2C slave.

AlessioR, 2020-05-23:
Thank you, it works.
I used a 40pin IDE connector with resistors soldered on it.
With long wires and no resistors it give me wrong results.

closed 2020-06-28 by guymcswain