joan2937/pigpio #354: What does "gpioWaveCreate: ERROR" mean

unicornis-pl, 2020-05-28:
Hi,
I get following error message when I call pi.wave_send_repeat(newWaveId):
_2020-05-28 23:16:08 myDoCommand: on=40 off=C delay=2
2020-05-28 23:16:08 myDoCommand: on=AA0 off=0 delay=2
2020-05-28 23:16:08 myDoCommand: on=8 off=0 delay=2
2020-05-28 23:16:08 myDoCommand: on=8 off=20 delay=200
2020-05-28 23:16:08 gpioWaveAddGeneric: numPulses=204 pulses=B666BE40
2020-05-28 23:16:13 gpioWaveCreate:
2020-05-28 23:16:13 gpioWaveCreate: ERROR wid=0 CBs 409=409 BOOL 204=308 TOOL 0=0
2020-05-28 23:16:13 gpioWaveCreate: Wave Stats: wid=0 CBs 409 BOOL 204 TOOL 0
2020-05-28 23:16:14 gpioWaveTxSend: wave_id=0 wave_mode=1_

What does BOOL 204=308 mean? I could find anything in documentation.

The error is generated by the sanity check in pigpio.c:
```
/* Sanity check. */

   if ( (numCB   != (CB-waveInfo[wid].botCB))    ||
        (numBOOL != (BOOL-waveInfo[wid].botOOL)) ||
        (numTOOL != (waveInfo[wid].topOOL-TOOL)) )
   {
      DBG(DBG_ALWAYS, "ERROR wid=%d CBs %d=%d BOOL %d=%d TOOL %d=%d", wid,
         numCB,   CB-waveInfo[wid].botCB,
         numBOOL, BOOL-waveInfo[wid].botOOL,
         numTOOL, waveInfo[wid].topOOL-TOOL);
   }
```
Do I do anyhing wrong defining the waveforms, so they fail the saninty check?

My program on RPi works more or less as expected, ocasionally (once every two-three days) totally freezing. That is why I started to look onto error logs. 

Thanks in advance for any hints how to start looking for the root cause.

joan2937, 2020-05-29:
You need to include a small but complete Python script which exhibits the problem.

unicornis-pl, 2020-05-29:
import pigpio
# pins controlled by pigpio
pins =  [0x0B,9,7,6,5,3,2]

pulses = [
    #   FEDCBA9876543210
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #23=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #22=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #21=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #20=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000100 , 0b0000000001001000 , 2], # Dat pulse bit #19=1
    [ 0b0000000001000100 , 0b0000000000001000 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #18=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #17=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #16=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000100 , 0b0000000001001000 , 2], # Dat pulse bit #15=1
    [ 0b0000000001000100 , 0b0000000000001000 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #14=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #13=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #12=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #11=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #10=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #9=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #8=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #7=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #6=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #5=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #4=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #3=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000100 , 0b0000000001001000 , 2], # Dat pulse bit #2=1
    [ 0b0000000001000100 , 0b0000000000001000 , 2], # Clk pulse
    [ 0b0000000000000100 , 0b0000000001001000 , 2], # Dat pulse bit #1=1
    [ 0b0000000001000100 , 0b0000000000001000 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #0=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000101010100000 , 0b0000000000000000 , 2], # All digits off
    [ 0b0000000000001000 , 0b0000000000000000 , 2], # Lat pulse
    [ 0b0000000000001000 , 0b0000000010000000 , 200], # Digit pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #23=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #22=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #21=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #20=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000100 , 0b0000000001001000 , 2], # Dat pulse bit #19=1
    [ 0b0000000001000100 , 0b0000000000001000 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #18=0
    [ 0b0000000001000000 , 0b0000000000001100 , 2], # Clk pulse
    [ 0b0000000000000000 , 0b0000000001001100 , 2], # Dat pulse bit #17=0
[... 135 lines cut ...]
    [ 0b0000000000000100 , 0b0000000001001000 , 2], # Dat pulse bit #0=1
    [ 0b0000000001000100 , 0b0000000000001000 , 2], # Clk pulse
    [ 0b0000101010100000 , 0b0000000000000000 , 2], # All digits off
    [ 0b0000000000001000 , 0b0000000000000000 , 2], # Lat pulse
    [ 0b0000000000001000 , 0b0000000000100000 , 200]# Digit pulse
]

#Pulses added:  204 Waveform len= 1200 us, uploaded in 46.12612724304199 ms
#New_wid 0

pi = pigpio.pi()
for pin in pins:
    pi.set_mode(pin, pigpio.OUTPUT)

waveform=[]
i = 1
for on,off,delay in pulses:
    print('#',i,'on=',format(on,'#018b'),'off=',format(off,'#018b'),'delay=',delay)
    waveform.append(pigpio.pulse(on, off, delay))
    i+=1
pi.wave_clear()
pulsesAdded = pi.wave_add_generic(waveform)  # add pulses to waveform
newWaveId = pi.wave_create()  # commit waveform
pi.wave_send_repeat(newWaveId)  # transmit waveform repeatedly
print('WaveID', newWaveId,'Pulses added: ', pulsesAdded, 'Waveform len=', pi.wave_get_micros(), 'us')

unicornis-pl, 2020-05-29:
I think I figured it out. Is it because the new pulse sets or clear the same bits as the previous one?
pulses = [
    #   FEDCBA9876543210
    [ 0b000000000**1**000000 , 0b000000000100**11**00 , 2], # Dat pulse bit #23=0
    [ 0b000000000**1**000000 , 0b000000000000**11**00 , 2], # Clk pulse

If so, can this error cause problems or it's just an info?

guymcswain, 2020-05-29:
> What does BOOL 204=308 mean? I could not find anything in documentation.

The BOOL, TOOL, as well as the CB are resources that get consumed when waves are added/created.  For the waveforms you are generating, BOOL is the number of On Off Level words stored in memory.  It looks like the tally of BOOL taken when the wave was added did not match after it was created.  It looks like you may be using version 76.  If this is the case, could you revert to an earlier version to see if the same behavior exists?

unicornis-pl, 2020-05-31:
Version 75 nor 71 does not report such error.

guymcswain, 2020-05-31:
@unicornis-pl ,
I have found a bug in v76 logic that optimized dma writes using a 2D stride.  The tally of resources is now correct and no error is reported.  However, I don't have a convenient way to verify your waveforms are being correctly generated.  Can you please checkout, compile and test the newly uploaded branch `issue_354` and test it in your environment?

unicornis-pl, 2020-06-01:
No errors reported using my waveforms:
```
pi@garaz-pi:~/pigpio-issue_354 $ cat /dev/pigerr
^C
pi@garaz-pi:~/pigpio-issue_354 $ pigpiod -v
7601
```

guymcswain, 2020-06-01:
Thank you for your help in reporting and verifying the issue.  Please let me know if you continue to see: 

> occasionally (once every two-three days) totally freezing

At this point, I'm assuming it is related to the bug identified.  I'm also a bit concerned why you didn't see the error report from the python client.  I don't use it much but I do remember a switch documented to turn errors on/off.

unicornis-pl, 2020-06-02:
So far so good, no issues.
These hangs were very strange, the Raspberry Pi froze completely, no ping, no ssh, nothing in the logs - no new entries in any log after the freeze. The ethernet LEDs were flashing, but the unit was completely unresponsive. Hard reset was the only option. As it runs as a headless I really couldn't figure out what was the root cause. 
Unfortunately I have not redirected /dev/pigerr to the text file, so I do not have the log for pigpiod after reset. Now I did so using cat /dev/pigerr > pigerr.txt & (hope it will do the trick)
Is it possible that the bug you found could block the raspbian completely?

guymcswain, 2020-06-02:
Quite possibly.  Since it dealt with the DMA it could have accessed areas of memory it shouldn't have.  Given the create Wave API returned an error I would thought the client would have stopped at that point and not have proceeded with the transmit of the wave.

unicornis-pl, 2020-06-03:
Thanks for the explanation. 

> Given the create Wave API returned an error I would have thought the client would have stopped at that point and not have proceeded with the transmit of the wave.

This error could not be detected in the Python. Part of the script that deals with it:

```
self.pi.wave_clear()
pulsesAdded = self.pi.wave_add_generic(self.waveform)  # pulsesAdded :204
newWaveId = self.pi.wave_create()  # newWaveId: 0
cbs = self.pi.wave_send_repeat(newWaveId)  # cbs: 405
```
Please note that none of the return values indicated an error condition. Error is thrown by wave_create, but it returns 0, which does not indicate an error.

unicornis-pl, 2020-06-10:
No errors for 8 days now.

PR #365 (2020-06-28): V77 release
Version 77 is a patch release for bugs and documentation clarifications.

Bug fixes:

- 5395bbe  fix #354 and fix #363.
- 328e09a  fix #350

Documentation:

- 1a98364  fix #355
- 1c5b27f  fix #356 
- d7379ba fix #341

Reverted commits:  (previously on develop branch)

- f51642a  "Remove excessive PROT_EXEC"
- a77351a  "Calling exit from a signal handler is not safe."
closed 2020-06-28 by guymcswain