joan2937/pigpio #355: PIGPIO event_cancel

eslavko, 2020-06-06:
Hello..
In pigpio event callback is bug/wrong documentation. Even example doesn't work.
_cb1 = pi.event_callback(22, cbf)
cb1.event_cancel()_

Throw error
_AttributeError: _event instance has no attribute 'event_cancel'_

The correct syntax should be:
_ cb1.cancel()_

guymcswain, 2020-06-06:
Please provide more information.

eslavko, 2020-06-07:
cb1.event_cancel() throws error.
in forum I got response that cb1.cancel() is correct syntax. So manual is wrong.

guymcswain, 2020-06-07:
I think you are referring to the documentation example under API `event_callback()` that shows the use of a non-existent method `event_cancel()`.  

This appears to be a documentation error.  Thank you.

eslavko, 2020-06-07:
Correct!

PR #365 (2020-06-28): V77 release
Version 77 is a patch release for bugs and documentation clarifications.

Bug fixes:

- 5395bbe  fix #354 and fix #363.
- 328e09a  fix #350

Documentation:

- 1a98364  fix #355
- 1c5b27f  fix #356 
- d7379ba fix #341

Reverted commits:  (previously on develop branch)

- f51642a  "Remove excessive PROT_EXEC"
- a77351a  "Calling exit from a signal handler is not safe."
closed 2020-06-28 by guymcswain