joan2937/pigpio #356: pi.wave_tx_at()  return wrong number

eslavko, 2020-06-06:
I found that pi.wave_tx_at() return wrong number.
While waveforms runs the reported number is sometime 9998. Sometime is even 66 and that was unexisting waveform.
I found that when running "pi.wave_chain([255, 0, test, 255, 3])" I got sometime 9998 but no other numbers.
While running pi.wave_chain([255, 0, test, 255, 1, 200, 200]) I got 9998 and other numbers too.

It's same on PI or remote.
Here is code to test. 

`#!/usr/bin/env python
import pigpio
import time

#connect to PI
pi = pigpio.pi('t0804k20.local')

if not pi.connected:
    exit()

pi.wave_clear()

test=[]
test.append(pigpio.pulse(1<<4, 0, 10))
test.append(pigpio.pulse(0, 1<<4, 10))
pi.wave_add_generic(test)
test = pi.wave_create()

print "test where 9998 apears where shouldn't"
pi.wave_chain([255, 0, test, 255, 3])
while 1:
    x=pi.wave_tx_at()
    if x!=test:
        print x
        break

pi.wave_tx_stop()

print "Test to show that other ids than correct or 9998 are returned"
pi.wave_chain([255, 0, test, 255, 1, 200, 200])
while 1:
    x=pi.wave_tx_at()
    if (x!=test) & (x!=9998):
        print x
        break

pi.wave_tx_stop()
pi.wave_clear()
`

guymcswain, 2020-06-06:
Are you seeing a wave id other than 9999, 9998 or test?  Since you are chaining waves, I assume it's possible to catch it at the completion when one might expect PI_WAVE_NOT_FOUND or PI_NO_TX_WAVE.

eslavko, 2020-06-07:
Yes.
When I run above script I got as result 66 to. And this wave is not even created. There are just only 1 wave numbered 0 created.
..I run pigpiod -x -1 -s 1   if that's matter.

eslavko, 2020-06-07:
infinite chain pi.wave_chain([255, 0, test, 255, 3])
should run forever and wave_tx_at() should return id=0 in that case, but throws 9998 too.

pi.wave_chain([255, 0, test, 255, 1, 200, 200]) is even worse.
it throws 0 (correct) and 9998 and 66 randomly sometime.

eslavko, 2020-06-07:
I observed that (wrong) number returned is the number of last wave EVER created +1.
So as I had used 65 waved before the 66 come from that.
To prove that there is procedure to follow:

reastart pigpiod daemon
run supplied script
run supplied script again

The 1st time the wrong ID is 1 in the 1st part and 2 in second part
after dummy is created.

when run 2nd time the both wrong IDs are 2.

I hope that this will help to discover what is wrong.

`#!/usr/bin/env python
import pigpio

#connect to PI
pi = pigpio.pi('t0804k20.local')

if not pi.connected:
    exit()

pi.wave_clear()

test=[]
test.append(pigpio.pulse(1<<4, 0, 10))
test.append(pigpio.pulse(0, 1<<4, 10))
pi.wave_add_generic(test)
test = pi.wave_create()

print ("test wave id is",test)
pi.wave_chain([255, 0, test, 255, 1, 250, 250])
while 1:
    x=pi.wave_tx_at()
    if x==9999:
        break
    if (x!=test) & (x!=9998):
        print x
pi.wave_tx_stop()

#creadte dummy wave but don't use it anywhere
dumy=[]
dumy.append(pigpio.pulse(1<<4, 0, 10))
dumy.append(pigpio.pulse(0, 1<<4, 10))
pi.wave_add_generic(dumy)
dumy = pi.wave_create()

print ("test wave id is",test)
pi.wave_chain([255, 0, test, 255, 1, 250, 250])
while 1:
    x=pi.wave_tx_at()
    if x==9999:
        break
    if (x!=test) & (x!=9998):
        print x
pi.wave_tx_stop()
pi.wave_clear()

raw_input('done')
`

guymcswain, 2020-06-07:
`if (x!=test) & (x!=9998):`
`if (x!=test) & (x!=9998):`
?

eslavko, 2020-06-07:
while 1:
    x=pi.wave_tx_at()
    if x==9999:break
    if (x!=test) & (x!=9998):
        print x,
print

gives
Test to show that other ids than correct or 9998 are returned
1 1 1 1 1 1 1 1 1 1 1
done

guymcswain, 2020-06-07:
I'm questioning the usage of a 'bitwise and' test between two inequality tests.  Shouldn't this be a 'logical and'.

eslavko, 2020-06-07:
in this case doesn't matter
the returned ID is 1 and that wave is not created at all! So here is something buggy

joan2937, 2020-06-07:
Thinking about it I'm not even sure this function works with wave chains.

eslavko, 2020-06-07:
As I observed that I can conclude that works buggy. 
I need to filter out 9998 and any ID with numbers over last created one.

guymcswain, 2020-06-07:
You need to fix your your conditional test using a logical and.

eslavko, 2020-06-07:
I do both! Same result!

eslavko, 2020-06-07:
..actualy I used that from pigs script and get same result.

eslavko, 2020-06-08:
> Thinking about it I'm not even sure this function works with wave chains.

What purpose is then to function exists. In all other ways only one wave is running and user should know the ID.

guymcswain, 2020-06-08:
Ok, I have verified your test script and per Joan's comment, will accept this as a documentation issue.

eslavko, 2020-06-09:
From my point of view this is not documentation issue. 
The wave can be started with 3 commands:
1.wave_send_once
2.wave_send_repeat
3.wave_chain

with 1 and 2 there can be only one wave. So asking "wave_tx_at" here is nonsense. But with wave started with 3 the asking "wave_tx_at" can be essential. And this doesn't work as should. This is clearly bug. Probably in time when wave is changed to next one or next loop the result is incorrect.

Workaround I get is:
Can't rely on 9998 report as is reported even if wave is transmitted. 
If in consecutive 2 or 3 reads you get 9998 then is probably true 9998.

Keep track what is highest wave ID created or what IDs are in current wave_chain. 
if reported ID is higher than last wave created (or not in current wave chain) its garbage. So read ID again until you got expected wave ID or 9999 or (at least twice in row) 9998.

Until now I didn't got wrong report twice in row.

guymcswain, 2020-06-25:
Can you retest with branch `issue_356`?

guymcswain, 2020-06-26:
> Can you retest with branch issue_356?

Please ignore this.  I believe I now understand why Joan said this API was not intended for chained waves.  `wave_tx_at()` takes a sample from the currently executing DMA address and then does a look-up into a table associating wave ids to DMA control block address ranges (indices actually).  Since wave chaining takes the DMA controller outside this table - executing delay loops for example - the resulting look-up value is outside the ranges in the table.  

I'm going to leave this as documentation clarification - "`wave_tx_at()` is not intended to be used with chained waves."

eslavko, 2020-06-26:
Yes. At least there should be some notification that function doesn't work, or what they did.

PR #365 (2020-06-28): V77 release
Version 77 is a patch release for bugs and documentation clarifications.

Bug fixes:

- 5395bbe  fix #354 and fix #363.
- 328e09a  fix #350

Documentation:

- 1a98364  fix #355
- 1c5b27f  fix #356 
- d7379ba fix #341

Reverted commits:  (previously on develop branch)

- f51642a  "Remove excessive PROT_EXEC"
- a77351a  "Calling exit from a signal handler is not safe."
closed 2020-06-28 by guymcswain