joan2937/pigpio #358: bb_serial_read_open doesn't clear receive buffer

eslavko, 2020-06-07:
Hello...
When issuing  "bb_serial_read_open(user_gpio, baud, bb_bits)" the receive buffer is not cleared.
I had issue in my app as I need to open/close that beacouse I run rx/tx with different baudrates.

when I get data I needed with "bb_serial_read(user_gpio)" I close receiver with "bb_serial_read_close(user_gpio)". 
But as there are lattency in mean time some other (garbage) bites are already received betwen read and close.

when I again open with "bb_serial_read_open(user_gpio, baud, bb_bits)" 
and read with "bb_serial_read(user_gpio)" I got that garbage bits too.
As bitrate is high and lattency is longer than few bits there is no way to know which byte is the correct one. 

I see solution that when  "bb_serial_read_open(user_gpio, baud, bb_bits)" or "bb_serial_read_close(user_gpio)" is issued the receive buffer should be cleared.

joan2937, 2020-06-07:
Are you sure?  The receive buffer is allocated from the stack in the open (and emptied) and free'd in the close.  I think you are seeing something else.

eslavko, 2020-06-07:
Actually I'm wrong here.
Indeed the buffer IS cleared when bb serial is opened. But there is hack!
As I do open bb serial in script (to have as low lattency as possible) I discovered that when serial is opened they capture data from past!
For test I do wave chain to transmit byte 1,2,3 and another with byte 4,5.
The pigs script start bb receiving when see waveform (4,5) started.
and I receive bytes 3,4,5 often, or even garbage as receiving seems to start somewhere in midbyte of byte 2 or 3. 

This is probably caused by some buffering and probably not solvable at all.
For finding that it take me few days. So please just add in manual that when receiver is started the data capture start around 200-400us in past!

eslavko, 2020-06-07:
With more deep observing I find that start point where capture begin is very random. As I observed for now it can be around half milisecond earlier or late.

guymcswain, 2020-06-07:
It will depend on the sampling rate and the buffer size specified for pigpiod.  This is not a bug.

eslavko, 2020-06-08:
Should be nice if that is in documentation.

closed 2020-06-08 by eslavko