joan2937/pigpio #360: PWM pin on CM3 has no hardware clock?

vkotarov, 2020-06-15:
Hello!

I am trying to get pigpio working on CM3 (CM3+ is also planned).
To my knowledge, GPIOs 2 - 45 should be safe to control on CM3, right?
I am trying to get hardware PWM running on GPIO12 (ALT0) with the following commands: 

1. Set GPIO mode:

```
cm3:~# pigs m 12 0
cm3:~# pigs mg 12  
4
cm3:~# 
```

So far it seems OK, GPIO12 set to ALT0 (PWM).

2. Set HW clock on GPIO12 to 5KHz:

```
cm3:~# pigs hc 12 5000
-94
ERROR: GPIO has no hardware clock
cm3:~# 
```

HWVER and last 5 lines of /proc/cpuinfo are below:

```
cm3:~# pigs hwver
10625184
cm3:~# tail -n 5 /proc/cpuinfo 

Hardware	: BCM2835
Revision	: a220a0
Serial		: 0000000012aafdd1
Model		: Raspberry Pi Compute Module 3 Rev 1.0
cm3:~# 
```

I am running kernel 4.19.93

```
cm3:~# uname -a
Linux cm3 4.19.93 #1 SMP Tue Apr 28 10:33:36 UTC 2020 armv7l GNU/Linux
cm3:~# 
```

joan2937, 2020-06-15:
GPIO 12 does not have a hardware CLK mode.

http://abyz.me.uk/rpi/pigpio/pigs.html#HC

You should set the PWM clock on GPIO 12.

http://abyz.me.uk/rpi/pigpio/pigs.html#HP

If you want a square wave on GPIO 12 at 5 kHz use

**pigs hp 12 5000 500000**

vkotarov, 2020-06-16:
Pfft ... my bad .. should've read the manual several times before posting here.

Best regards,
Vesselin.

closed 2020-06-16 by vkotarov