joan2937/pigpio #361: Set glitch and set filter seem to be ignored

CuriousJames, 2020-06-18:
Hello,

This is likely a poor implementation issue, but I don't see what I'm doing wrong from the documentation.

I have a VERY bouncy button, so I want to handle that and saw that PiGPIO should do that for me, which would be great!

I've tried setting the glitch noise filters both before and after setting up a pi.callback (both edge) but these filters appear to be ignored - i.e. it's still just as bouncy, even if I set it to something stupid like 30000 (ms) it bounces several times in a second.  I've also tried setting the filter before and after the Pull Ups, but no change.

Could it be that the callback doesn't respect the noise and glitch filters?  In which case what should I use instead to know when it's up or down after the filter?

Here's my code snippet from reader.py from this git (I haven't committed this snippet on account of this issue, but feel free to clone my git and give it a go with the following snippet):

    pi.set_noise_filter(p.pins["doorbellButton"],30000,3000)

    # register these GPIO pins to run cbf on rising or falling edge
    global cb1,cb2,cb3,cb4
    cb1 = pi.callback(p.pins["doorStrike"], pigpio.EITHER_EDGE, cbf)
    cb2 = pi.callback(p.pins["doorbell12"], pigpio.EITHER_EDGE, cbf)
    
    cb3 = pi.callback(p.pins["doorbellButton"], pigpio.EITHER_EDGE, cbf)
    cb4 = pi.callback(p.pins["doorSensor"], pigpio.EITHER_EDGE, cbf)
    pi.set_pull_up_down(p.pins["doorbellButton"], pigpio.PUD_UP)
    pi.set_pull_up_down(p.pins["doorSensor"], pigpio.PUD_UP)

My Repo is here: https://github.com/CuriousJames/DIYAC

P.S.
I appear to be running v71 of PiGPIO, can I upgrade that via apt-get if that is the problem?

Thanks very much, James.

guymcswain, 2020-06-18:
> 30000 (ms)

The units are microseconds so 30 msec is not a crazy value to use when dealing with buttons.  I have not used the `set_noise_filter` api so not sure if Joan intended it to be used as a debounce filter.

CuriousJames, 2020-06-18:
Thnks so much, I definitely misread microseconds as miliseconds!  Sorry.

Can you tell me the easiest way to update to the latest version though please?

Thanks, James.

joan2937, 2020-06-18:
For the latest version try http://abyz.me.uk/rpi/pigpio/download.html

The noise filter was not intended for debounce (use the glitch filter for debounce).

The noise filter was really intended for use with radio transmitters which do a lot of static followed by quiet followed by signal proper.

CuriousJames, 2020-06-18:
Gotcha, I only tried the noise one after the glitch one didn't appear to work, I've gone ahead and used glitch and it's working perfectly, thanks!

Also thanks for that link to install the latest version!

closed 2020-06-18 by CuriousJames