joan2937/pigpio #362: Revision codes changed in pi kernel

zinen, 2020-06-18:
I kept getting this error when trying to compile this code on my Raspberry Pi 3 Model B rev 1.2. (64bit).
```
Unhandled signal 11
```
 A lot of searching and I started digging into this repo. I found that CPU amount was not set to 4 cores, but to 1 core. I think this is due to the way function `gpioHardwareRevision()` in the file `pigpio.c` is detecting my board version, since my BCM2837 chip is responding as it is a BCM2835 chip when running `cat /proc/cpuinfo`.

I then found this quote from the Raspberry Pi team specifically on the revision codes:
`As of the 4.9 kernel, all Pis report BCM2835, even those with BCM2836, BCM2837 and BCM2711 processors. You should not use this string to detect the processor.` Quote from [here](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/revision-codes/README.md) written 2. aug. 2019 12.53 CEST.

More readning at (stackExchange)[https://raspberrypi.stackexchange.com/questions/75089/bcm2835-on-an-rpi-3]

guymcswain, 2020-06-18:
What version of the library are you using?
64-bit OS has never been officially supported, but that doesn't mean it shouldn't work - *caveat utilitor*.

zinen, 2020-06-18:
Running pigpiod-v76 on OS Alpine-v3.12. Did a compile too on my 'Raspberry Pi 1 Model B Rev 2', same version of pigpiod and same Alpine version with no problems at all.

But as the Quote above says, all new kernels are reportning back as BCM2835 (aka chipset from Raspberry Pi 1).

guymcswain, 2020-06-18:
Yes, but since v75 (perhaps earlier) pigpio checks both `/proc/cpuinfo` for legacy as well as `the new location` that is recommended by Raspbian.  So I'm surprised this is failing.  I'll get back to you on 'the new location' so you can verify if Alpine is compliant.

guymcswain, 2020-06-18:
Give this a shot:

> Stop the currently running instance of pigpio.
> From a working directory, run:
> sudo pigpiod -c4 -g 2> pigpiod.log
> Allow a few seconds to complete initialization, then Cntrl-C (SIGINT) to abort.
> Run
> head pigpiod.log
> If there was any error in the hardware revision decode it should appear in this output.

zinen, 2020-06-18:
Ran it, here is hoping you can make sense of this:

*I did remove some of the log file, as it contained a lot of the same errors*
```
2020-06-18 19:43:16 gpioInitialise: not initialised, initialising
2020-06-18 19:43:16 initInitialise: 
2020-06-18 19:43:16 gpioHardwareRevision: 
revision=a22082pi_peri_phys=3f000000pi_dram_bus=c0000000pi_mem_flag=42020-06-18 19:43:16 initClearGlobals: 
2020-06-18 19:43:16 initClearGlobals: f0 is 8000
2020-06-18 19:43:16 initClearGlobals: f1 is 4000
2020-06-18 19:43:16 initClearGlobals: f2 is 2000
2020-06-18 19:43:16 initCheckPermitted: 
2020-06-18 19:43:16 initPeripherals: DMA #14 @ 7FA75C1E00
2020-06-18 19:43:16 initPeripherals: debug reg is 10028E00
2020-06-18 19:43:16 initAllocDMAMem: bmillis=120 mics=5 bblk=20 bcyc=1600
2020-06-18 19:43:16 initMboxBlock: init mbox zaps failed
```

Also did a `hexdump /proc/device-tree/system/linux,revision`, my result was this 
```
0000000 a200 8220                              
0000004
```

guymcswain, 2020-06-18:
Good, so this is saying BCM2837 is detected - and more importantly there were no errors reported.  The peripheral addresses should be the same for all revisions except Pi4 (BCM2711).  But I'll double check...

At this point, I don't see a cause and effect between you getting `unhandled signal 11` and the cpu revision.

guymcswain, 2020-06-18:
> I did remove some of the log file, as it contained a lot of the same errors

Maybe you should post these errors.

zinen, 2020-06-18:
Added to post above https://github.com/joan2937/pigpio/issues/362#issuecomment-646275083

guymcswain, 2020-06-18:
Are you running 32bit os?  Must be if the same ran on pi1 but thought I'd double check.

zinen, 2020-06-18:
Running `uname -a` reveals that it is running on 64bit architecture (aarch64 below). Also tried installing Alpine-v3.11.6 to test this also, but i get the same error as with Alpine-v3.12.
```
Linux c475146e5b2e 4.19.71 #1 SMP PREEMPT Fri Jan 31 09:53:40 UTC 2020 aarch64 Linux
```
Since Alpine doesn't have graphical interface i tried running `pigpiod -c4 -g 2 -a 1> pigpiod.log` (with -a 1) as i remembered reading something about GPU earlier. No luck, same error.

guymcswain, 2020-06-18:
I think I've reached my limit of usefulness in helping you.  If you want I can flag this issue with 'help wanted' and 'alpine 64bit' or similar to see if someone in the community can assist.

guymcswain, 2020-06-19:
I was recently reminded of this hack:  [bind mount a faux variant over /proc/cpuinfo](https://www.raspberrypi.org/forums/viewtopic.php?p=1428987#p1428987)

You can try it if you believe alpine-64 is reading/relying on the faulty information contained in `/proc/cpuinfo`.

zinen, 2020-06-20:
> I was recently reminded of this hack: [bind mount a faux variant over /proc/cpuinfo](https://www.raspberrypi.org/forums/viewtopic.php?p=1428987#p1428987)
> 
> You can try it if you believe alpine-64 is reading/relying on the faulty information contained in `/proc/cpuinfo`.

I'll test that, and also test running Alpine 32bit to see if its the 64bit that make the problem.

zinen, 2020-06-20:
Reinstalled Alpine-v3.12 but as 32-bit and not 64-bit, now everything works.

The OS running 64-bit might have been the problem. As i can see there are multible issues around this. I'll look into the others to see if i can snif up a fix.

[Issues with label 64-bit](https://github.com/joan2937/pigpio/issues?q=label%3A64-bit+)

bakpakin, 2020-06-21:
Possibly related, but I also got a SIGSEGV when running on 64 bit archlinuxarm. I tracked down the issue with valgrind:

```
==10270== Invalid read of size 4
==10270==    at 0x49D1338: fclose@@GLIBC_2.17 (in /usr/lib/libc-2.31.so)
==10270==    by 0x4EF53AB: gpioHardwareRevision.part.0 (pigpio.c:13733)
==10270==    by 0x4F0CA8B: initInitialise (pigpio.c:8246)
==10270==    by 0x4F0EAFB: gpioInitialise (pigpio.c:8668)
==10270==    by 0x4EF0F2B: _janet_init (pigpiowrap.c:75)
==10270==    by 0x1344DB: janet_core_native (in /usr/local/bin/janet)
==10270==    by 0x13B04F: run_vm (in /usr/local/bin/janet)
==10270==    by 0x13CD8B: janet_continue_no_check (in /usr/local/bin/janet)
==10270==    by 0x1169EB: main (in /usr/local/bin/janet)
```

The problem seems to be calling `fclose` on NULL (line 13733 in pigpio.c):

```
if (rev == 0)
   {
      DBG(DBG_USER, "searching /proc/device-tree for revision");
      filp = fopen ("/proc/device-tree/system/linux,revision", "r");

      if (filp != NULL)
      {
         uint32_t tmp;
         if (fread(&tmp,1 , 4, filp) == 4)
         {
            /*
               for some reason the value returned by reading
               this /proc entry seems to be big endian,
               convert it.
            */
            rev = ntohl(tmp);
            rev &= 0xFFFFFF; /* mask out warranty bit */
         }
      }
      fclose(filp);
   }
```

The fix is moving fclose inside the if statement.

However, this is only hit because there was a permission error - the program needs root permissions. Not to mention other issues on archlinux arm I hit after fixing this one.

guymcswain, 2020-06-23:
@zinen  - Have you looked into PR #342?  It is related to 64-bit OS as well.

@bakpakin  - Thanks for the finding the `fclose` issue.  For this and all other 64-bit contributions, we rely on those contributors to test and validate the 64-bit functionality.  We only test on 32-bit systems to make sure nothing breaks in that world.

zinen, 2020-06-23:
> @zinen - Have you looked into PR #342? It is related to 64-bit OS as well.
>
Haven't seen that but it looks promising.  I'll test it.

zinen, 2020-06-23:
Tried PR #342 with semi succes.  Only worked when I ran `pigpiod -g` (without `-a 1`) as mentioned in PR. 
It worked and pigpiod ran on Alpine 64-bit, but only the first time i fired up pigpiod. When i killed it and tried to run it again a second time I was back to this error:
```
initMboxBlock: init mbox zaps failed
```
A reboot of the Pi fixed it, and I can run the service again one time before the error. Seams as PR #342 have a problem releasing memory or something.

guymcswain, 2020-06-24:
**pigpio** was written for a 32-bit userland.  It is possible to run pigpio successfully on a machine with a 64-bit kernel and a 32-bit userland as is done on Gentoo 64.  If Alpine with 64-bit kernel is running 64-bit userland then pigpio is unlikely to work.

I think a command you can run to show the 'bit-ness' of your installed userland is: `getconf  LONG_BIT`.

zinen, 2020-06-24:
*I just figured if multiple users wanted a working 64-bit version, the best way was to share details of it here. Maybe someone can figure it out.*

When running `getconf LONG_BIT` i get 64. So yeah its a 64-bit userland.

guymcswain, 2020-06-24:
> if multiple users wanted a working 64-bit version, the best way was to share details of it here

I'm not disagreeing.  But what I think is needed for that to work is a comprehensive review and conversion of pigpio to work in both 32-bit and 64-bit environments.  For example, there are many instances of variables declared as `int` that are assumed to be 32 bits in size.  Up until now, the author, Joan, has been reluctant to wade into 64 bit without an official OS from the Raspberry Pi org.  With the recent introduction of [Raspberry Pi OS (64 bit) beta test version](https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370&sid=632aef420ec573532a172536e7291b1a) now may be the time.

I would be happy to host an experimental 64-bit branch on this repo for the community to post pull requests with patches to make this conversion.

zinen, 2020-06-24:
That's not a bad idea to combine all PR and issues into a 64-bit compatibility development branch. Maybe also write in README.md about this branch to pipe new PR/issues into that branch...

alex-giusi-tiri, 2020-07-01:
I too would like to add a note, that I am using NixOS aarch64 (with 64-bit userland (LONG_BIT==64)), and I get this error and `pigpiod` terminates shortly:
```
$ sudo time pigpiod -b 10000 -g -s 10;
2020-06-30 23:52:46 sigHandler: Unhandled signal 11, terminating

Command exited with non-zero status 255
1.29user 27.04system 0:28.54elapsed 99%CPU (0avgtext+0avgdata 109048maxresident)k
0inputs+0outputs (0major+80342minor)pagefaults 0swaps
```

Thank you!

guymcswain, 2020-07-01:
I must ask why you are using `time` to initiate the daemon.

alex-giusi-tiri, 2020-07-01:
> I must ask why you are using `time` to initiate the daemon.

Only to show how long it takes until it terminates. :-)

guymcswain, 2020-07-01:
Please enable the debug output: 
```
sudo pigpiod -c4 -g 2> pigpiod.log
```

guymcswain, 2020-07-02:
Update:  The `fclose` issue identified by @bakpakin has been patched in the `develop` branch.  This branch feeds into the main branch on every new release of pigpio.  Please retest with the `develop` branch.

alex-giusi-tiri, 2020-07-05:
> Please enable the debug output:
> 
> ```
> sudo pigpiod -c4 -g 2> pigpiod.log
> ```

> Update: The `fclose` issue identified by @bakpakin has been patched in the `develop` branch. This branch feeds into the main branch on every new release of pigpio. Please retest with the `develop` branch.

I have tried running [0] `pigpiod` for a few minutes, and it seems to be working(?) (I have attached [1] the log file :-)):

I tried issuing [0] `pigs`, while [0] `pigpiod`  was running, while being logged in as root, but:
```
# pigs write 26 1;
socket connect failed
```
I didn't notice any change in the log's output when I ran this command, corresponding to the attempt of the connection.

I also noticed that issuing Control+C, for stopping [0] `pigpiod`, triggers this; I am unsure if that is expected:
```
2020-07-05 17:06:48 sigHandler: Unhandled signal 2, terminating

2020-07-05 17:06:48 gpioTerminate: 
```

Thank you for your support! :-)

[0] I used `pigpio`, from the "develop" branch, from the commitment https://github.com/joan2937/pigpio/commit/487bf29673f13f96f85673ffcafbabfb2da29ee1, in the same aarch64 64-bit userspace environment
[1] [pigpiod-487bf29673f13f96f85673ffcafbabfb2da29ee1-0.log](https://github.com/joan2937/pigpio/files/4875683/pigpiod-487bf29673f13f96f85673ffcafbabfb2da29ee1-0.log)

guymcswain, 2020-07-06:
> socket connect failed

It never connected so the daemon via the socket.  The log shows no evidence of an attempted/failed connection.  From the log file it looks like it never got out of the DMA initialization process.  So while it made further progress it is still fundamentally broke in this area for 64-bit.

> I also noticed that issuing Control+C, for stopping [0] pigpiod, triggers

The daemon process ignores all signals.  The way to stop it is to kill the process (from [pigpio faq](http://abyz.me.uk/rpi/pigpio/faq.html) ):
```
Check that it is running with the command
ps aux | grep pigpiod

Kill the daemon with

sudo killall pigpiod

If your own program is acting as the daemon it may be removed as follows.

Find its process id (pid).

cat /var/run/pigpio.pid

Kill the program with

sudo kill -9 pid

If the above doesn't work do the following and try starting the daemon again
sudo rm /var/run/pigpio.pid
```

guymcswain, 2020-07-06:
As an experiment, can you start the daemon with an alternate dma initialization method?   Try `sudo pigpiod -a 1` 

Include the other options as before to generate a log file.
Thanks.

alex-giusi-tiri, 2020-07-06:
Thanks for the suggestion! Here [0] it is. It ran for about 45 seconds. It seems to me that it could not pass past the DMA initialization phase. :-(

Thanks, again! :-)

[0] [pigpiod-487bf29673f13f96f85673ffcafbabfb2da29ee1-2.log](https://github.com/joan2937/pigpio/files/4876459/pigpiod-487bf29673f13f96f85673ffcafbabfb2da29ee1-2.log)

guymcswain, 2020-07-06:
@alex-giusi-tiri ,
Please try another test, this time we bypass the DMA initialization on startup of pigpiod.  This allows to see what issues may lie beyond DMA for 64 bit environment.  (Meanwhile, I will attempt to debug the previous log files you provided from DMA init.)

- Download new branch `aarch64user`.  We'll use this branch for all 64-bit experiments.

- Run `sudo pigpiod -mgc4 2> somefile.log`

- In another terminal try some pigs commands: `pigs pigpv`, etc.

guymcswain, 2020-07-10:
Finally got Raspberry Pi OS 64 installed on RPI3B+
```bash
pi@raspberrypi:~/Downloads/pigpio-aarch64user $ uname -a
Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux
pi@raspberrypi:~/Downloads/pigpio-aarch64user $ getconf LONG_BIT
64
```
To my amazement - thinking there would be significantly more work to do - all pigpio tests are passing.  (Maybe I'm not really running 64-bit, lol.)

joan2937, 2020-07-10:
@guymcswain One of the Raspbian guys (plugwash) did the 64-bit work on pigpio.  You'll see comments sprinkled around the code.

alex-giusi-tiri, 2020-07-12:
> @alex-giusi-tiri ,
> Please try another test, this time we bypass the DMA initialization on startup of pigpiod. This allows to see what issues may lie beyond DMA for 64 bit environment. (Meanwhile, I [...]

> Finally got Raspberry Pi OS 64 installed on RPI3B+
> 
> ```shell
> pi@raspberrypi:~/Downloads/pigpio-aarch64user $ uname -a
> Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux
> pi@raspberrypi:~/Downloads/pigpio-aarch64user $ getconf LONG_BIT
> 64
> ```
> 
> To my amazement - thinking there would be significantly more work to do - all pigpio tests are passing. (Maybe I'm not really running 64-bit, lol.)
> 
> Edit: I reverted commit [2ffcae2](https://github.com/joan2937/pigpio/commit/2ffcae2960802cb23f22e51adc99b9f7ed1104af) (see above) on my local repo, recompiled and pigpio is still happy under RPiOS64.

@guymcswain,
Please excuse my delayed reply; I was rather busy with work this week, and I got some chance to work on this today...
I obtained pigpio-[aarch64](https://github.com/joan2937/pigpio/tree/aarch64user)-[2ffcae2960802cb23f22e51adc99b9f7ed1104af](https://github.com/joan2937/pigpio/commit/2ffcae2960802cb23f22e51adc99b9f7ed1104af) and ran as root

pigpiod -mgc4 2>[pigpio-2ffcae2960802cb23f22e51adc99b9f7ed1104af-0.log](https://github.com/joan2937/pigpio/files/4907978/pigpio-2ffcae2960802cb23f22e51adc99b9f7ed1104af-0.log);

and tried some `pigs w` commands as root, and, to my amazement, as well, it worked! The LEDs turned on!

Thank you!

alex-giusi-tiri, 2020-07-12:
**Update**

This time I tried, but it errored out:
 ```
pigpiod -b 10000 -g -s 10;
2020-07-12 00:14:02 sigHandler: Unhandled signal 11, terminating
```

Now, I am trying, but it is continuously saying things like, while still running:
```
pigpiod -b 10000 -c 4 -g -s 10;
[...]
2020-07-12 00:24:31 initZaps: pf50=A1000000000226DF
2020-07-12 00:24:31 initZaps: pf51=A1000000000226E0
2020-07-12 00:24:31 initZaps: pf52=A1000000000226E1
2020-07-12 00:24:31 initPagemapBlock: block=406
2020-07-12 00:24:31 initZaps:
2020-07-12 00:24:31 initZaps: pf0=A10000000002254F
2020-07-12 00:24:31 initZaps: pf1=A100000000022550
2020-07-12 00:24:31 initZaps: pf2=A1000000000225F5
[...]
```

Should I stop it? Would it finish?

Thanks!

alex-giusi-tiri, 2020-07-12:
It finished: [pigpio-2ffcae2960802cb23f22e51adc99b9f7ed1104af-2.log](https://github.com/joan2937/pigpio/files/4908010/pigpio-2ffcae2960802cb23f22e51adc99b9f7ed1104af-2.log)

alex-giusi-tiri, 2020-07-12:
`pigpiod -b 10000 -g -m -s 10` does not show any output, but `pigs w` work.

guymcswain, 2020-07-12:
> I think, perhaps that it didn't work before, because I was using the NixOS' Linux kernel instead of the RaspberryPi's one.

This very well may be the problem.  Without the proper drivers to talk to the VC core via mailboxes and such, it appears the DMA memory allocations are not getting setup correctly.  Btw, the -m option on the pigpiod command in this branch is hacked to completely skip over all DMA setup.  So a 'successful' run with the -m again points to a problem with your kernel not being able to correctly allocate shared memory for DMA.

guymcswain, 2020-07-12:
@zinen , I think your OS - Alpine-v3.12(64-bit) - must be built from the **raspberry pi linux** in order for pigpio to work.

alex-giusi-tiri, 2020-07-12:
I was using the RaspberryPi kernel for the last four comments, and I was using it previously as well, only that previously, I was using both, alternately, so I wasn't paying too much attention to which one I was using on that boot, but now I had realized that it may matter, since the NixOS kernel doesn't detect the DS18B20 temperature sensor... The -m seems to help on the RaspberryPi kernel as well, since it doesn't work without it, on it too... :-?

guymcswain, 2020-07-12:
Well, I'm fairly ignorant when it comes to building linux kernels so take my comments with a grain of salt.  But from my earlier post, the 64-bit raspberry pi os runs pigpio across its complete test suite.  So I feel pretty confident in stating that your kernel is missing some important components.

alex-giusi-tiri, 2020-07-12:
Indeed, porting the RaspberryPi Linux changes into the mainline kernel is a work-in-progress, as far as I understand. [0] Some things, like the RPi overlays are not compatible with the mainline device tree files, so it may very well be such incompatibilities...
I think, maybe, that there might be some more work which needs to be done to make a 64-bit version, like you said before...

[0] [/boot/config.txt related questions for Raspberry Pi](https://github.com/NixOS/nixpkgs/issues/67792#issuecomment-542403125) (Point 2)

closed 2020-12-19 by guymcswain