joan2937/pigpio #363: Wave corruption

eslavko, 2020-06-22:
I found this unexpected behaviour of wave.
I do generate two square waves one of 50Hz and other 100Hz as one wave.
Then I create wave twice with same data (doesn't same efect with different)
and when I play the 1st generated wave they came out corrupted and second one
came out as should. I do capture real output from PI. There is hardware low pass filter so wave looks distorted but that doesn't matter.
On pin4 I have power switch for analog extension but that shouldn't affect result.

`#!/usr/bin/env python
import pigpio
import time

PIN1=18
PIN2=15

pi = pigpio.pi('t0804k20.local')
pi.set_mode(PIN1, pigpio.OUTPUT)
pi.set_mode(PIN2, pigpio.OUTPUT)
pi.hardware_clock(4, 100000) # 100 KHz clock on GPIO 4. I need that for external watchdog

#create waveform with two square waves.
#One with 20ms period and other at 40ms period as complex wave
w=[]
w.append(pigpio.pulse((1<<PIN2), (1<<PIN1), 10000))
w.append(pigpio.pulse((1<<PIN1), 0, 10000))
w.append(pigpio.pulse(0, (1<<PIN1)|(1<<PIN2), 10000))
w.append(pigpio.pulse((1<<PIN1), 0, 10000))

#clear any previous data
pi.wave_tx_stop()
pi.wave_clear()

#create wave W1
pi.wave_add_generic(w)
W1 = pi.wave_create()

#create wave W2 (with same data)
pi.wave_add_generic(w)
W2 = pi.wave_create()

#Play 1st generated data. This came out CORRUPTED
pi.wave_send_repeat(W1)
time.sleep(4)

#Plat 2nd generated data. Came out perfect
pi.wave_send_repeat(W2)
time.sleep(4)

#stop generation
pi.wave_tx_stop()
pi.wave_clear()`

[image]
[image]

eslavko, 2020-06-22:
I do another tests.
When wave is playing with "wave_send_repeat" just generation of other wave can corrupt allready running one. And there is more on that. The shape of new wave (or pins used) determine if waveform is corrupted or not. 
just try that program and uncomment one line where pulse is added to q. The corruption doesn't appear on all of them!

`#!/usr/bin/env python
import pigpio
import time

PIN1=18
PIN2=15

pi = pigpio.pi('t0804k20.local')
pi.set_mode(PIN1, pigpio.OUTPUT)
pi.set_mode(PIN2, pigpio.OUTPUT)
pi.hardware_clock(4, 100000) # 100 KHz clock on GPIO 4. I need that for external watchdog

#create waveform with two square waves.
#One with 20ms period and other at 40ms period as complex wave
w=[]
w.append(pigpio.pulse((1<<PIN2), (1<<PIN1), 10000))
w.append(pigpio.pulse((1<<PIN1), 0, 10000))
w.append(pigpio.pulse(0, (1<<PIN1)|(1<<PIN2), 10000))
w.append(pigpio.pulse((1<<PIN1), 0, 10000))

#clear any previous data
pi.wave_tx_stop()
pi.wave_clear()

#create wave W1 and start
pi.wave_add_generic(w)
W1 = pi.wave_create()
pi.wave_send_repeat(W1)
time.sleep(2)

#after 4 second playing (perfect wave) just create new one (don't start it)
q=[]

#just uncomment one of those lines
#------------------
q.append(pigpio.pulse((1<<PIN2), (1<<PIN1), 10000))     #does corupt
#q.append(pigpio.pulse(0, (1<<PIN1), 10000))            #doesn't corrupt
#q.append(pigpio.pulse((1<<PIN2),0, 10000))             #does corupt
#------------------

pi.wave_add_generic(q)
W2 = pi.wave_create()  	 #that's corrupt W1 playing
time.sleep(2)

#stop generation
pi.wave_tx_stop()
pi.wave_clear()

`

eslavko, 2020-06-22:
This issue is relleated to version 76! 
On version 75 or earlier is OK.

guymcswain, 2020-06-22:
Please test with branch `issue_354`

eslavko, 2020-06-23:
my example works with _issue_354_

PR #365 (2020-06-28): V77 release
Version 77 is a patch release for bugs and documentation clarifications.

Bug fixes:

- 5395bbe  fix #354 and fix #363.
- 328e09a  fix #350

Documentation:

- 1a98364  fix #355
- 1c5b27f  fix #356 
- d7379ba fix #341

Reverted commits:  (previously on develop branch)

- f51642a  "Remove excessive PROT_EXEC"
- a77351a  "Calling exit from a signal handler is not safe."
closed 2020-06-28 by guymcswain