joan2937/pigpio #364: Python cleanup 'pi.write' not working

CuriousJames, 2020-06-27:
Hello,

I'm using the 'atexit' module on python in order to cleanup GPIO before my python program exits.

As it's a Pi that opens a public door I very much want it to close the door before the python program potentially shuts down for a long time (unexpectedly).

Unfortunately pi.write is the only function I've found that doesn't work in this cleanup function, it must have something to do with python cleaning up on it's own side (I assume) as pi.write works inside the program, just not in this cleanup function I made.

> #
> # cleanup
> # makes things clean at exit
> #
> def cleanup():
>     l.log("DBUG", "cleanup started")
> 
>     # close the door
>     try:
>         # outH.setDoor("closed")
>         l.log("DBUG", "attempting close")
>         pi.write(27, 0)
>     except Exception as e:
>         l.log("WARN", "Unable to close the door", e)
>     # release gpio resources
>     try:
>         pi.stop()
>     except Exception as e:
>         l.log("WARN", "Unable to stop PiGPIO conenction", e)
>         pass
>     # stuff for a nice clean exit
>     atexit.register(cleanup)

When I catch the error python says:
> 'NoneType' object has no attribute 'send'

When I don't catch the error python says:
> Error in atexit._run_exitfuncs:
> Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/pigpio.py", line 993, in _pigpio_command
>     sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
> AttributeError: 'NoneType' object has no attribute 'send'

If anyone has any technique for me to ensure a GPO is set to low before the python program closes I'd really appreciate that!  Be it a fix for this code I've written, or a different method entirely.

Thanks, James.

guymcswain, 2020-06-27:
It looks like the network socket has closed.

Questions:
1.  If you know you are about to  exiting, why not cleaning then?
2.  Does Python throw an exception (before it crashes) that you can trap to run cleanup before exiting?

CuriousJames, 2020-06-28:
Ahh I see.

1: This is what that function is designed to do - but I'm not clear on where this is executed on the python's script's lifecycle - clearly it's already cleared down some things before hitting this function, which isn't ideal.
2: I've found another Python module now ('signal') which does the same thing (allows you to register a function or functions to execute when an interupt or other stop/pause signal is sent to the script, and there I'm still able to do PiGPIO calls.  So this is resolved now.
3: Yes, localhost.

closed 2020-06-28 by CuriousJames