joan2937/pigpio #368: Unexpected Diagnostic Message on Closing PiGPIO

dhrynyk, 2020-07-05:
Hi:

In closing off PiGPIO I received the following, presumably, internal diagnostic message. I thought it may be helpful to pass the information along to you.

Exiting...

#####################################################
pigpio version=71 internals=200
micros=5 allocMode=0 dmaInitCbs=2 DMARestarts=1
samples 0 maxSamples 0 maxEmit 0 emitFrags 0
cbTicks 61141, cbCalls 344815
pipe: good 0, short 0, would block 0
alertTicks 344804, lateTicks 0, moreToDo 10
     1559         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         2        40       206       390     89082   2650340     89169       382       186        39         1         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         0         1 
#####################################################

Addition information:
- Device: rpi4b
- OS: Buster
- Device: Single I2C VEML7700

Please let me know if you would like any additional information. 

David

guymcswain, 2020-07-07:
From `gpioTerminate()`?

joan2937, 2020-07-07:
Thanks for the report.  The check for DMA restarts was added in (I think) when the multi-core Pis came out and DMA addressing was changed.  DMA was restated in your run which is why the message was printed.

Given that we aren't flooded with similar reports this isn't really indicating a practical problem, it's more of a curiosity.

closed 2020-09-29 by guymcswain