joan2937/pigpio #369: hardware_clock issue on version 75

adrienmo, 2020-07-14:
Hello!

I am trying to use the command `hardware_clock` of pigpiod (v75), I want to set the pin 4 of my raspberry pi0W to 13.5 Mhz.
Following the documentation I just execute `hardware_clock(4, 13500000)`, but using a scope I am only seeing a 800 Hz signal. Also it seems it makes my pi0 totally unstable and the UART start not to behave properly.

I used `get_PWM_frequency` to verify the frequency set and I can read an ok value (`13499974`)

Is it a bug from the library? Did I forget to call something else first?

Thanks

guymcswain, 2020-07-14:
Do you have an HDMI monitor connected or otherwise have audio enabled?  Which OS and what model are you running on?

adrienmo, 2020-07-14:
@guymcswain no hdmi monitor connected, I am not sure for the audio. I am using nerves framework, which has the bare minimum (based on buildroot) : 
- https://www.nerves-project.org/
- https://github.com/nerves-project/nerves_system_br

guymcswain, 2020-07-14:
So this is a special Linux build.  Pigpio is heavily dependent on the raspberry pi Linux.  Have you tried running the pigpio test suite - or at least the python test `x_pigpio.py`?

adrienmo, 2020-07-17:
Hello @guymcswain , I tried with raspbian + python and it gives the same result, so it does not seem related to the linux distribution.
I was able to make it work using this sample code directly: http://abyz.me.uk/rpi/pigpio/code/minimal_clk.zip

guymcswain, 2020-08-31:
@adrienmo ,
I have loaded version 75 on a raspberry pi zero W and run the api hardware_clock(4, 13500000) which results in 13.5Mhz square wave observed on my oscilloscope on GPIO 4.  I can only conclude that you have a configuration issue without more information.

closed 2020-09-29 by guymcswain