joan2937/pigpio #370: pigpiod, on start, interferes with GPIOs in use by other devices

iovsiann, 2020-07-23:
Hi, thank you for the making and supporting this excellent library. Joan, you are my open source hero.

I tried running pigpiod to control servos with a microphone array also connected to the Raspberry Pi 4B's expansion slot. The servos use GPIO 12 and 13. The microphone array (Respeaker) uses some other remaining GPIOs for I2S, SPI and I2C. Running "sudo pigpiod" (from the command line) breaks the microphone audio capture. The servos work well with pigpiod when the mic array is disconnected - and vice versa, the mic array works well unless I run "sudo pigpiod"

I tried Gordon's wiringpi instead of pigpiod to control GPIOs 12 and 13 using hardware PWM - wiringpi works well and does not interfere with other GPIOs (I.e. I can capture audio and control servos simultaneously).

I also tried "sudo pigpiod -x 4096" to mask out all GPIOs except GPIO 12, "sudo pigpiod -x 0" to mask out all GPIOs - running any of those commands still messes up the microphone array.

Any advice would be much appreciated.
Thank you

guymcswain, 2020-07-23:
Try `sudo pigpiod -t 0`

iovsiann, 2020-07-23:
Guy, awesome, that worked - thank you for the instant answer!

closed 2020-07-23 by iovsiann