joan2937/pigpio #372: SPI SCLK frequency is double the baud rate on Pi 4B (pigpio v1.71)

alanxyzhou, 2020-07-23:
I've noticed that if I set the baud rate with spiOpen to x, I measure a frequency of 2x on the SCLK line. For example, setting baud to 100000 gives me a 200kHz clock. I am wondering if anyone else has noticed this. 

I am just continually reading the MISO line:

    int handle = spiOpen(0, 100000, 0);
    char buf[1];
    for (;;) { spiRead(handle, buf, 1); };

guymcswain, 2020-07-24:
Rpi model, pigpio version, is audio in use or HDMI monitor attached?

alanxyzhou, 2020-07-24:
Raspberry PI Model 4B
pigpio 1.71
No audio or HDMI, but VNC server and SSH are enabled for headless connection

guymcswain, 2020-07-24:
> No audio or HDMI

Good, HDMI will enable audio.
> pigpio 1.71

I'm expecting a two digit number, 71 maybe?

guymcswain, 2020-07-24:
I'm not sure what VNC does to audio.  Just to be sure can you put `gpioCfgClock(5, 1)` before `gpioInitialise()` in your script.

alanxyzhou, 2020-07-24:
> I'm expecting a two digit number, 71 maybe?

My apologies. I am on pigpio 77.

> Just to be sure can you put gpioCfgClock(5, 1) before gpioInitialise() in your script.

Tried this; no change. 

> I'm not sure what VNC does to audio.

Can I ask how the audio would affect this? Maybe I can look for ways to disable audio entirely.

alanxyzhou, 2020-07-24:
I also tried disabling VNC, but that did not fix the issue.

guymcswain, 2020-07-24:
> Can I ask how the audio would affect SPI?

I'm not sure about SPI if you are using the peripheral - not the pigpio bit-banged SPI.  In general, however, there are many conflicts between audio and pipgio.  Digital audio uses the PCM peripheral which is the default source of DMA timing for pigpio.  Analog audio uses the PWM peripheral which pigpio uses for servo, pwm and waveforms.

I'll try this on my Pi4 in the next few days to see if I can reproduce.

guymcswain, 2020-07-25:
I'm seeing 100KHz as measured using `setAlertFunc()` callbacks on SCLK transitions.

alanxyzhou, 2020-07-27:
Interestingly, if I use setAlertFunc to time the rising edges of SCLK, I also see 100kHz.

But you can see clearly from this oscilloscope measurement that the clock (blue line) is actually 200kHz. 

[image]

guymcswain, 2020-07-27:
Interesting.  I'll have to relocate my pi4 so that I can put a scope probe on it.

guymcswain, 2020-07-28:
I set this up on an Pi ZeroW that was easily at hand.  The first burst of 8 bits is clocked at 100KHz but subsequent bursts are clocked at a higher rate - 160KHz in the case of Pi0W.  I feel better that the `setAlertFunc()` wasn't lying to me as I was only measuring the first byte transfer.  However, why the bit rate changes is a mystery to me.

@alanxyzhou , can you verify on your scope if the baud rate of the initial byte transfer is correct and only for subsequent transfers the bit clock changes to 200KHz (presumably in the case of Pi4B)?

@joan2937 , can you provide any insight as to why the SPI peripheral baud rate is different from what is initialized in the `spiOpen()` api?

alanxyzhou, 2020-07-28:
I couldn't verify if only the first pulse was correct. I was only seeing the clock go straight into 200kHz.

FWIW, gpioHardwareClock always outputs the correct timing.

guymcswain, 2020-07-28:
Edit your /boot/config.txt with the line: `core_freq=250`.   Reboot then retest please.

alanxyzhou, 2020-07-29:
Ah, I think I understand the issue after reading this doc: https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

The Pi4B specifically has a default core_freq of 500, exactly double the expected 250. Which explains the exactly doubled frequency. 

I don't know if there is a way for pigpio to detect which platform it is running on, but it would seem that if the 250000000 value here https://github.com/joan2937/pigpio/blob/master/pigpio.c (line 4550) can change to 500000000 on the 4B, the issue is resolved. 
Or better yet, detect core_freq and use that value instead of the hardcoded 250MHz. 

I've been just dividing my desired baud rate by 2 when initializing SPI as a workaround.

guymcswain, 2020-07-29:
> I don't know if there is a way for pigpio to detect which platform it is running on, but it would seem that if the 250000000 value here https://github.com/joan2937/pigpio/blob/master/pigpio.c (line 4550) can change to 500000000 on the 4B, the issue is resolved.

The problem is that some models do 'vpu frequency scaling' which affects *core_freq* which affects both SPI and UART timing.  To keep timing consistent the core frequency must be locked.  Setting `core_freq=250` in config.txt works on all models except Pi4.  [This thread, github.com/raspberrypi/firmware/issues/1308,](https://github.com/raspberrypi/firmware/issues/1308) discuss this very same issue and suggests a future enhancement  to lock frequency on Pi4 may be forthcoming.

We currently distinguish Pi4 models from the others and could easily modify the line of code you suggest.  However, I'd like to see how this gets resolved in the enhanced feature discussed above.  If you would like to patch your own version of pigpio change [this line](https://github.com/joan2937/pigpio/blob/a2997c58a3837fc8b76df7948d0b4ce81e3bf99c/pigpio.c#L4550) to : 
```c
if (pi_is_2711)    spiReg[SPI_CLK] = 500000000/speed;
else    spiReg[SPI_CLK] = 250000000/speed;
```
You will still need to lock your Pi4's core frequency as outlined in the referenced thread above.

alanxyzhou, 2020-07-29:
Thank you for that thread, it was quite informative. 

In my case, I had `enable_uart=1` in my /boot/config.txt already -- I've confirmed that this line alone was enough to lock core_freq to 500MHz, without touching `core_freq` or `core_freq_min`. I will use these workarounds until something more permanent is implemented. 

Thanks for the help!

hk298, 2023-03-11:
I'm running pigpio 79 and came across the same issue. The baud rate was twice the rate that I specified in spiOpen. Shouldn't this bit of code that Guy posted three years ago be part of the official version?
> ```c
> if (pi_is_2711)    spiReg[SPI_CLK] = 500000000/speed;
> else    spiReg[SPI_CLK] = 250000000/speed;
> ```