joan2937/pigpio #373: pigpiod_if2.c

darkyellow, 2020-07-29:
The code here

int notify_pause(int pi, unsigned handle)
   {return pigpio_command(pi, PI_CMD_NB, handle, 0, 1);}

Should it not be PI_CMD_NP rather than PI_CMD_NB?

guymcswain, 2020-07-29:
From just looking at the source you posted, it would seem so.  I'll accept a PR if you can test to verify it is indeed broken, fix and retest showing working.  Thanks in advance.

darkyellow, 2020-07-29:
unfortunately I am pretty new to the library so just starting out and finding my way around so I don't have any use case to test with. I was just looking through all the functions and noticed the above.

If you see in pigpio.c 

 case PI_CMD_NP: res = gpioNotifyPause(p[1]); break;

(line 2199)

This can never be called from the remote process because of the mismatch.

guymcswain, 2020-07-29:
If you plan to write some code requiring this, I can add a patch to the development branch.  If all's well then it will get picked up in the next release.

joan2937, 2020-07-29:
I think I just decided that since Pause is equivalent to Begin with bits 0 are identical I'd just use Begin.  It's the same in the Python.  More of an oddity rather than an error.

darkyellow, 2020-07-30:
Thanks, makes sense. I found another

int i2c_process_call(int pi, unsigned handle, unsigned reg, uint32_t val)

and

int i2c_block_process_call(int pi, unsigned handle, unsigned reg, char *buf, unsigned count)

both use PI_CMD_I2CPK and PI_CMD_I2CPC is not used anywhere. I would guess there is similar valid reasoning here too?

joan2937, 2020-07-30:
@darkyellow 

No, that is an error.  The code for i2c_process_call should be.

The code should be
```c
int i2c_process_call(int pi, unsigned handle, unsigned reg, uint32_t val)
{
   gpioExtent_t ext[1];

   /*
   p1=handle
   p2=reg
   p3=4
   ## extension ##
   uint32_t val
   */

   ext[0].size = sizeof(uint32_t);
   ext[0].ptr = &val;

   return pigpio_command_ext
      (pi, PI_CMD_I2CPC, handle, reg, 4, 1, ext, 1);
}
```
In this case Python seems to be using the correct call.

closed 2020-09-29 by guymcswain