joan2937/pigpio #374: Change argument type of pigpio_start from `char*` to `const char*`

dawidowski, 2020-07-30:
Currently, function `pigpio_start` takes arguments only as 'char*'
I suggest to change function parameters types to `const char*`, so it could also accept`char*` and string literals. This will make interactions with STL `const char* std::string::c_str()` method easier.

This would also require change arguments type of `pigpioOpenSocket` what should not be a problem, because it passes these arguments directly to POSIX `getaddrinfo`.

joan2937, 2020-07-30:
Will this change break any existing code?

dawidowski, 2020-07-30:
This change would require a little modification in `pigpio_start` function, but it would also change signatures of this one and `pigpioOpenSocket` functions, hence I assume it would involve chagnes in test scritps, code etc.

joan2937, 2020-07-30:
By break I mean does anyone using the code have to make any changes to their code?  If so the change would break existing code.

dawidowski, 2020-07-30:
Sorry, now I get what you have meant.

`pigpio_start` with arguments `char*` are more strict for what can be passed into it. Changing it to `const char*` makes it more flexible.

After modification `pigpio_start` should still accept `char *` variables as arguments without any problems.

guymcswain, 2020-08-29:
Please submit a pull request including test cases for review.

dawidowski, 2020-09-17:
Do you have any recommendation where is should put test for `start_pigpio` in `x_pigpiod_if` and `x_pigpiod_if2`? Testsuits from 1 to 9 seems to be related to different actions. `start_pigpio` is not tested anywhere and only used in `main()`.

guymcswain, 2020-09-17:
Without having looked at the test code, I would just suggest a test  (or test set) is run after each variant of `pigpio_start` and ending with `pigpio_stop`.

guymcswain, 2021-03-02:
Fixed in v79

closed 2021-03-02 by guymcswain