joan2937/pigpio #375: Operation not permitted

HanDaSeul, 2020-08-04:
Using rpi 4B 8GB, ubuntu 20.04, and pigpio is v77. 

It said: initPeripherals: mmap gpio failed (Operation not permitted)
When I try ps aux | grep pigpiod, it said: ubuntu      3957  0.0  0.0   2828   668 pts/1    S+   01:31   0:00 grep --color=auto pigpiod
And then, when i try killing it, it said: pigpiod: no process found

guymcswain, 2020-08-04:
If this is a 64 bit OS, it is likely missing some of the raspberry pi Linux bits needed to initialize the shared memory for DMA.  I haven't seen any 64 bit OS, outside of raspberry pi os, work running pigpio  in userland.

juergh, 2020-08-05:
The gpio library is trying to mmap using PROT_EXEC but recent distros mount /dev with the noexec flag so that won't fly. Is there any valid reason why PROT_EXEC is required? I see that commit ad52dcdc834c2f813c32002c26f82e00a9f0ac5f removed that flag and then commit f51642ad1374b5bc6bfcf076c7459f7a7bf0db54 reverted that change again. No information given in either commit so who knows what's going on?

guymcswain, 2020-08-05:
@juergh , you are correct - I didn't know what was going on wrt PROT_EXEC.  A PR was submitted which seemed ok, ie didn't break the tests, but was later challenged by another user but the original author never responded.  Furthermore, it appeared to be related to 64-bit which has other complications.  

I'm happy to make available test patches to the develop branch - but we don't support non raspberry pi OS in our testing so we rely on community support/feedback.

juergh, 2020-08-06:
@guymcswain Just created a PR to drop the flag. A simple call of gpioInitialise() on Raspberry Pi OS 64 bit works just fine. If dropping that flag causes problems in other places then those places need fixing. IMO, there should be zero need for PROT_EXEC for this library.

FWIW, the change to mount /dev with 'noexec' went into initramfs-tools 0.135. Buster (which Raspberry Pi OS is currently based on) is on 0.133. The next Debian release will use >0.135.

hugo-a-garcia, 2020-09-23:
I have not tested the development branch yet BUT i did remove PROT_EXEC from a copy of master, compiled it and ran it on both a Pi4 with Ubuntu 64bit and a Pi3 with Raspian Pi OS. They communicate properly.

hugo-a-garcia, 2020-09-23:
I tested the develop branch of pigpio with two Pi's connected via CAN where the CAN modules use SPI and one of the Pi's is connected to an ultrasound sensor using pigpio. The setup works.

The test code can be found here:
https://github.com/bluecorn/raspberry-can-HC-SR04
https://github.com/bluecorn/raspberry-can-master-ultrasound

lefrog, 2020-09-29:
I've tested removing `PROT_EXEC` from master branch and running it on a Raspberry PI 2 Model B on Ubuntu 20.04 32bit and works like charm. All the tests from `x_pigpio` ran successfully blinking my LED on GPIO25 :)

closed 2020-09-29 by guymcswain

smurfix, 2023-12-11:
OK, so why has the removal of PROT_EXEC since been reverted?  @guymcswain can you explain?

juergh, 2023-12-12:
Where do you see that? It's still gone:
https://github.com/joan2937/pigpio/blob/master/pigpio.c#L7338