joan2937/pigpio #380: Script Operations

MRIIOT, 2020-08-24:
Using Python [callback](http://abyz.me.uk/rpi/pigpio/python.html#callback) function, I am watching rising edge of pin 22.

Using default pigpiod timing the callback is invoked every 1 second.
`tag 999 w 22 1 mils 1000 w 22 0 mics 5 jp 999`

Removing the delay after pullling pin 22 low, the callback is sporadic at best.
`tag 999 w 22 1 mils 1000 w 22 0 jp 999`

Would the usage of WAIT be correct here or is a delay of a single pigpiod cycle in above working script the correct approach?
`tag 999 w 22 1 mils 1000 w 22 0 wait 0x200000 jp 999`

Thanks
Chris

joan2937, 2020-08-25:
Why jp (jump positive).  I think you should be using an unconditional jmp (jump).

guymcswain, 2020-08-31:
Marking this with 'no action' and will close unless an issue can be identified.

closed 2020-09-29 by guymcswain