joan2937/pigpio #382: feature request: increase number of allowed loops in gpioWaveChain

korroktheslavemaster, 2020-09-09:
Right now there is a hard limit of 20 loops (as per documentation) in a wave chain
I am trying to create a ramp pulse input with ~100 steps of increasing frequencies (5000us to 10us), however the library only allows 20 steps at most. Is this a hardware limitation or can it be changed?
In the code I can see the limit is defined as 
```
#define WCB_COUNTERS (WCB_CNT_PER_PAGE * PI_WAVE_COUNT_PAGES)
```
where `WCB_CNT_PER_PAGE` is 2 and `PI_WAVE_COUNT_PAGES` is 10
Any help would be appreciated.

guymcswain, 2020-09-10:
Where are you seeing this limitation?  The documentation, as I read it, shows a loop count of 
```
loop x + y*256 times
```
is possible.  Both x, y are 8-bit integers.

korroktheslavemaster, 2020-09-10:
sorry i meant loop **counters**, not loop count
```
The code is currently dimensioned to support a chain with roughly 600 entries and 20 loop counters.
```
i.e. i can have only 20 loop blocks in the chain, having more throws `PI_CHAIN_COUNTER`

guymcswain, 2020-09-10:
Ok, thanks for clarifying.  The limitation may be due to the allocated dma control blocks in memory which may have been arbitrary when @joan2937 implemented the code.  Even if it is, to change it could require a fairly involved effort.

korroktheslavemaster, 2020-09-10:
Thank you. I'm up for contributing if @joan2937 thinks this is worthwhile, or I'd like to know if there's another way around this.

korroktheslavemaster, 2020-09-15:
I couldn't find any documentation related to sending synchronous wave chains. Do you mean using `wave_send_using_mode(PIN, pigpio.WAVE_MODE_ONE_SHOT_SYNC)` to send individual waves?

guymcswain, 2020-09-15:
Yes.  But now that I think about it, you would have to send each individual wave.  If they are 20ms long and your network isn't slow, you should be able to keep up using `gpiowavetxat` to know when the next wave is playing.