joan2937/pigpio #384: [observation] Docker and Chromium/Vivaldi kiosk hangs

MRIIOT, 2020-09-15:
I enjoy the flexibility and reliability of the pigpio library so I decided to move away from RPi.GPIO in our project.  The RPI4 loads the containers and boots into Chromium or Vivaldi  (I have found Vivaldi to be more responsive).  During testing I have experienced system hangs, browser crashes, and other oddities like no PWM output on a pin even though `pigs GDC` and `pigs PRF` reported correct values from within the container's shell.

I then narrowed down the strange behavior was only present when the browser was loaded by X.  System performed reliabily over reboots when the browser was removed from `/etc/xdg/openbox/autostart`.

Here is a sample setup reproducing the behavior.  Observations include system hangs, browser crashes, or pin 6 showing no signal.

https://github.com/MRIIOT/pigpio-docker

Thanks again for the awesome library.

guymcswain, 2020-09-15:
> I then narrowed down the strange behavior was only present when the browser was loaded by X.

If X mucks around with audio - such as setting PCM clock/timing - it may be conflicting with pigpio.  Try moving pigpio clock to alternate - see FAQ related to issues with sound.  HTH.

guymcswain, 2020-10-12:
I really can't help much with your environment since I know very little about Docker and have no experience running a browser alongside pigpio (and I'm unclear how you are using pigpio within this environment).  All I can offer are some suggestions/observations:

- Observation:  You are using the -a1 option to start the pigpio daemon.  Was there a problem using the default auto mode?  If so, this hints at a possible conflict with DMA channels.  Try using different DMA channels using the `-d` and `-e` options.
- Browsers use/enable audio so read the pigpio FAQ about issues with sound: Try `-t 0` option when invoking the daemon.
- Turn off audio.  Uncomment `audio=on` in config.txt.  Run `aplay -l` to see if audio drivers are installed.

closed 2020-12-19 by guymcswain