joan2937/pigpio #385: Initial falling edge detection not working`

QNimbus, 2020-09-17:
My issue seems related to #359 

However I can very easily AND reliably reproduce the issue using the method described below. First a description of my environment:

### Hardware:

- Raspberry Pi Zero WH (BCM 2835 SOC)
- GPIO pin 18 is used in my code below (but issue can be reproduced using any GPIO pin)
- Generated pulses either manually or using an old [rotary dial](https://www.google.com/imgres?imgurl=https%3A%2F%2Fupload.wikimedia.org%2Fwikipedia%2Fcommons%2F4%2F43%2FKiesschijf_T-65.jpg&imgrefurl=https%3A%2F%2Fnl.wikipedia.org%2Fwiki%2FKiesschijf&tbnid=XYKWofnTueWKFM&vet=12ahUKEwjGt4C-0O_rAhXOwKQKHSVSDfcQMyguegUIARD0AQ..i&docid=DTMcBdhfOnXyHM&w=1068&h=1068&q=t-65%20rotary%20dial&ved=2ahUKEwjGt4C-0O_rAhXOwKQKHSVSDfcQMyguegUIARD0AQ)

### Environment:

- `uname -a` -> Linux raspberrypi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
- `python --version` -> Python 2.7.16
- `python3 --version` -> Python 3.7.3
- `pigpiod -v` -> 77

### My testing code:

```python
#!/usr/bin/env python
# -*- coding: UTF-8 -*-

# test.py

# Low pulse lasts approximately 36ms
# High pulse lasts approximately 60ms
# Last low pulse lasts approximately 15ms
# Period is 96ms
# Frequency ~ 10.41hz
# Rise time is approximately 7-15µS
# Fall time is approximately 4-10µS

def detectEdgeCB(gpio, level, tick):
    print("detectEdgeCB " + str(level) + " " + str(tick))

if __name__ == "__main__":

    import time
    import pigpio

    PWM_GPIO = 18

    pi = pigpio.pi()

    if not pi.connected:
        print("Not connected to PIGPIO Daemon")
        exit()

    # GPIO as input
    pi.set_mode(PWM_GPIO, pigpio.INPUT)

    # Rise time ~20 µS and fall time ~15 µS - so 100 µS should be sufficient for software debouncing
    pi.set_glitch_filter(PWM_GPIO, 100)

    # GPIO pull-up
    pi.set_pull_up_down(PWM_GPIO, pigpio.PUD_UP)

    # Detect initial rotary dial turn
    detectEdge = pi.callback(
        PWM_GPIO, pigpio.EITHER_EDGE, detectEdgeCB)

    print("Waiting for rotary dial input (press CTRL-C to exit)...")

    try:
        while True:
            print("Direct " + str(pi.read(PWM_GPIO)) +
                  " " + str(pi.get_current_tick()))
            time.sleep(1)
    except KeyboardInterrupt:
        pass

    pi.stop()
```

### Output 1st time (i.e. after reboot or power on):

```
Waiting for rotary dial input (press CTRL-C to exit)...
Direct 1 1287979995
Direct 1 1288985335
Direct 1 1289989956
Direct 0 1290994696
Direct 0 1291999280
detectEdgeCB 1 1293000384
Direct 1 1293004731
detectEdgeCB 0 1293058644
detectEdgeCB 1 1293094359
detectEdgeCB 0 1293152389
detectEdgeCB 1 1293188055
detectEdgeCB 0 1293246115
detectEdgeCB 1 1293281870
detectEdgeCB 0 1293340370
detectEdgeCB 1 1293376310
detectEdgeCB 0 1293434575
detectEdgeCB 1 1293470740
detectEdgeCB 0 1293528895
detectEdgeCB 1 1293564930
detectEdgeCB 0 1293623106
detectEdgeCB 1 1293659411
detectEdgeCB 0 1293717871
detectEdgeCB 1 1293753926
detectEdgeCB 0 1293812566
detectEdgeCB 1 1293848481
detectEdgeCB 0 1293907691
detectEdgeCB 1 1293922766
Direct 1 1294010372
Direct 1 1295014399
Direct 1 1296018785
```
[image]

As you can see the first falling edge is not detected. (4th line in the output when GPIO goes from '1' to '0')

### Output every subsequent time (success):

```
Waiting for rotary dial input (press CTRL-C to exit)...
Direct 1 1497855392
Direct 1 1498860295
detectEdgeCB 0 1499649992
Direct 0 1499864555
Direct 0 1500869033
detectEdgeCB 1 1501437677
detectEdgeCB 0 1501496587
detectEdgeCB 1 1501532962
detectEdgeCB 0 1501591842
detectEdgeCB 1 1501627607
detectEdgeCB 0 1501685797
detectEdgeCB 1 1501721812
detectEdgeCB 0 1501780392
detectEdgeCB 1 1501816392
Direct 1 1501871616
detectEdgeCB 0 1501875148
detectEdgeCB 1 1501911108
detectEdgeCB 0 1501970128
detectEdgeCB 1 1502006298
detectEdgeCB 0 1502065343
detectEdgeCB 1 1502101573
detectEdgeCB 0 1502160568
detectEdgeCB 1 1502196918
detectEdgeCB 0 1502255719
detectEdgeCB 1 1502291789
detectEdgeCB 0 1502351079
detectEdgeCB 1 1502366119
Direct 1 1502876439
Direct 1 1503880759
Direct 1 1504885353
```

Observation: When executing the test code for the first time (i.e. after a reboot/power on) - the first falling edge is _never_ detected. I can also get the same behavior without having to reboot by executing `pigs pud 18 u; pigs pud 18 d; pigs pud 18 u; pigs pud 18 d`. After running this command the first run of my code will _not_ detect the 1st falling edge and every subsequent run will _always_ detect the 1st falling edge.

I managed to find a **workaround** which might make sense to the author of the `pigpio` daemon. If I include the code below and manually trigger a short pulse - the test code will always (correctly) detect the 1st falling edge. Even on the first run.

```
    if not pi.connected:
        print("Not connected to PIGPIO Daemon")
        exit()

    # ...
    # ^--- existing code 

    # GPIO as output
    pi.set_mode(PWM_GPIO, pigpio.OUTPUT)

    # Pulse for debugging
    pi.gpio_trigger(PWM_GPIO, 100, 1)
    time.sleep(1)

    # existing code --->
    # ...

    # GPIO as input
    pi.set_mode(PWM_GPIO, pigpio.INPUT)
```

### Recap

Using the Python library to interface with `pigpiod` on my Raspberry Pi I cannot detect the 1st falling edge on my GPIO pin when my script is first run. On every subsequent script run the 1st falling edge gets detected correctly and everything works fine. I've been able to _reliably_ and _consistently_ reproduce this behavior. 

After a reboot _or_ when manually pulsing the GPIO pin on the command line (`pigs pud 18 u; pigs pud 18 d; pigs pud 18 u; pigs pud 18 d`) the first execution of my script will miss the 1st falling edge again. And all subsequent script executions pick up the 1st falling edge without a problem.

Hope someone is able to reproduce this as well and maybe provide a fix (or a reason why my work around works?) If anyone needs any additional information I will be happy to provide it.

Thanks!

guymcswain, 2020-09-17:
Please generate the input signal from script as I don't have a rotary dial.  Also, try removing the glitch filter, or enable it after you are sure the input signal has stablelized.

guymcswain, 2020-10-12:
@QNimbus ,
Can you confirm that `pi.callback()` catches the initial edge (after power-on or reboot) if the glitch filter is removed?

lidodvr224, 2021-01-12:
12Jan21
Same issue for me.  I fell back to:

import RPi.GPIO as GPIO
GPIO.add_event_detect(gpioinput, GPIO.BOTH,
                          callback=interrupt_handler0,
                          bouncetime=int(maxbouncetime/10))

and it worked fine.  I'll try the workaround, as I prefer to use pigpio.
thanks

guymcswain, 2021-01-12:
Your feedback is appreciated.  Please let me know what you mean by 'workaround'.  My position is that you can't put a glitch filter - which requires two edges - in front of a piece of code that is expecting to catch the initial edge (after library is initialized).

lidodvr224, 2021-01-12:
I tried removing the glitch filter and it made no difference.  I also tried
the 'workaround' and it did not work, still intermittently missing the
first Falling Edge.

guymcswain, 2021-01-12:
Please post a sample of your code.

lidodvr224, 2021-01-12:
will do, pretty simple, as it's test code for something else.
it's pretty much the same as the original developer who noticed the anomaly
first, using BOTH EDGEs as trigger.   I also tested using just FALLING_EDGE
and got the same result. verified voltage levels using o-scope. RPi 3,
newest s/w.

lidodvr224, 2021-01-13:
##test_pig.py (this is NOT production code, for testing missing first FALLING_EDGE using pigpio!)

#/usr/bin/env python3

#test_pig.py (from startup_pig.py)

#torun: python test_pig.py
#toggle switch between <gpioinput> and 1K resistor to ground
#gpioinput is PULL_UP
#initial state: switch is in OFF position

#NOTE: PIGPIO misses FIRST FALLING_EDGE when switch is placed in ON position
#ON = ground here

#USES PIG (callback parameters = 3, (channel, level, tick)
#python test_pig.py

import time
import signal
import sys
import datetime
import subprocess, os

gpioflag = False
use_pig = True

if use_pig:
    import pigpio
    pig = pigpio.pi()

if not use_pig:
    gpioflag = True
    import RPi.GPIO as GPIO
    GPIO.setwarnings(False)
    GPIO.setmode(GPIO.BCM)   #use GPIO naming convention
    GPIO.cleanup()

    maxbouncetime = 265   #
    import RPi.GPIO as GPIO
else:
    maxbouncetime = 25000*8 # 

gpioinput  = 26  #monitor this input for ON or OFF
gpiooutput = 19  #turn on LED if 26=ON(to ground)

ON_FLAG0 = False  #global
debounce = 0.5

##########################################################

def open0(onflag):  #do this when interrupt occurs, ON state
        #falling_edge (3.3v to gnd)

        print("ON ON ON")
        onflag = True

        #turn on gpiooutput led
        if use_pig:
            pig.write(gpiooutput, True)
        else:
            GPIO.output(gpiooutput, True)

        return(onflag)  #for close0

def close0(onflag):        
        #rising_edge, gnd to 3.3v on gpio
        
        print("OFF OFF OFF")
                       
        #turn OFF gpiooutput led
[... 145 lines cut ...]
else:  #initially ON
    print("call interrupt_handler0")
    if use_pig:
        interrupt_handler0(gpioinput,0,0) #pin,FALLING_EDGE,tick
    else:
        interrupt_handler0(gpioinput)
    time.sleep(1)
    print("HW, initially ON")

time.sleep(1)

print("WaitingForInterrupt from gpio")
print(" ")

try:
    while True:
        signal.signal(signal.SIGINT, handler)
        #wait on interrupt_state0, interrupt_state1
        
        time.sleep(1)
except KeyboardInterrupt:
    print("Got cntrl+c")
    pass

``

guymcswain, 2021-01-15:
This code is confusing with multiple libraries and flags for what should be a simple test to determine if an input edge can be detected reliably.  I am frankly too busy to want to sort through it.

Your problem symptoms don't match the original authors issue description:  'Initial falling edge not working.'  If you read through the issue he describes in great detail that it is an initial condition problem but otherwise works reliably.  You indicated that you miss the edge intermittently.  I would recommend you tidy up your code and debug it further.

Please note that you cannot rely on reading an input from within the callback as the level may have changed.  That is why a level argument is supplied to the callback.  You should assess whether an edge occurs solely based on the level argument.

FreezeFiver, 2021-02-02:
First, appreciate your effort in building and maintaining this lib. It's awesome.

Now relating to this issue: I've been having the same issue, and could **reproduce it reliably**. HOWEVER, as I put together this post, I managed to learn more about what's going on. Basically:
1. Before utilizing `pigpio` in script, if a pin's value was at `0`, either because that's the default value since boot up (exp. `BCM 17`), or because it has been manually set to `0` (exp. set to `OUTPUT` mode), and ...
2. In script, during setup, call `set_glitch_filter` **after** `set_pull_up_down`;

THEN, when one starts listening for edge callbacks, the **1st fall-edge event will be missing**.

## Example
Say I wire up `BCM 17` to `GND`, with a button in between (a simple open/close switch):
```
 __/  __      <-- Simple Button
|       |
17    GND
```

Power up Pi,  ssh in, and run `python test.py` with the following code:
```python3
import pigpio

pi = pigpio.pi()

# read initial level from gpio 17
print(f'initial: level = {pi.read(17)}')

# set up gpio 17
pi.set_mode(17, mode=pigpio.INPUT)
pi.set_pull_up_down(17, pud=pigpio.PUD_UP)
pi.set_glitch_filter(17, steady=1000)

# read post-setup level from 17
print(f'postset: level = {pi.read(17)}')

# set up callback
def on_edge(pin, level, tick):
    print(f'on_edge: level = {level}')
pi.callback(17, edge=pigpio.EITHER_EDGE, func=on_edge)

# pause until exit
try:
    import signal
    signal.pause()
except KeyboardInterrupt:
    exit()
```

Then if I do a couple of clicks on the button (press down, and then release up), I'll get this:
```
initial: level = 0
postset: level = 1

on_edge: level = 1        < - - - I click once (down and up), and only get a SINGLE callback, missing the first "level = 0"

on_edge: level = 0        < - - - I click again (down and up) -- from now on, all clicks are working with 2 callbacks
on_edge: level = 1

on_edge: level = 0        < - - - and so on...
on_edge: level = 1
```

### Key Facts
* If I stop the script, and then run again, the callbacks would continue to work properly.
* If I set the pin level back to 0 (exp. in command line with `pigz` or `gpio`), and run the script again, the 1st fall-edge will be missing again.
* If I use a pin with initial value at `1` (exp. `BCM 7`) before running the script, there won't be an issue.

## The "Workaround"
I'm guessing the issue has something to do with the "glitch filter".

If I put `set_glitch_filter` **before** `set_pull_up_down`, the issue will be gone -- although if I start listening for callback **right after** setup (as in example code above), I'd immediately get a `level=1` edge event, before I press any button -- I'd assume that's from pulling up the pin from 0 to 1.

I hope the above makes sense, and would appreciate some feedback from dev on wether my guess was correct.

guymcswain, 2021-02-13:
Thanks for providing the detail in your analysis.  I've been busy with other matters but I will get back to you when I have more time to look at this again.  A couple of comments for now:

- You are correct that an edge will be produced upon setting the pull resistor if it is different than the power-on pull states - gpio 1-7 are pull-up and gpio >7 are pull-down after power-on reset.
- The glitch filter, as written, requires two edges to produce an output.  This also seems to trip people up looking for a notification on the first edge of an input.  It is possible for the glitch code to change but I've been reluctant to do so until I can establish the root cause of this issue.

lidodvr224, 2021-02-18:
good info (gpio 1-7 are pull-up and gpio >7 are pull-down after power-on
reset).  I was setting gpio input pin #26 to Pull-Up, and missing the first
'on' state of my analog toggle switch.  After reconfiguring to gpio input
pin #4,  the script works everytime.

systemdarena, 2021-04-16:
I've run in to this issue as well, and coding around it is quite annoying. I see no reason that a "glitch filter" as described in the docs would need to see 2 edges before edge detection starts working...

guymcswain, 2021-04-16:
> It is possible for the glitch code to change but I've been reluctant to do so until I can establish the root cause of this issue.

It's annoying to me when people don't follow up on the issue they've posted.  Your input is noted.

systemdarena, 2021-04-16:
Sorry, not sure I follow, would you have preferred me to open a separate issue? Or I can post more info here, but I am pretty sure my issue is exactly the same as the origin reporter's issue. If you want me to post any code examples from my tests let me know. I'd be happy to put in some more effort on my side. I appreciate the awesome library you have. It has been otherwise easy to use and worked well.

guymcswain, 2021-04-16:
No another issue is not necessary.  I'll remove the documentation label and fix initializing the glitch filter when I get around to it.  If you want something sooner you are welcome to submit a PR.  (I was expressing some frustration on not hearing back from OP.)