joan2937/pigpio #386: Deprecation of the sysfs interface and its effect on pigpio

kobuki, 2020-09-18:
According to the [Linux kernel documentation page](https://www.kernel.org/doc/Documentation/ABI/obsolete/sysfs-gpio):

>  This ABI is deprecated and will be removed after 2020. It is
>  replaced with the GPIO character device.

How does it affect the pigpio library? Is it going to be ported to the new interface?

guymcswain, 2020-09-18:
A quick search for `/sys/class` shows there are five APIs affected - all relating to "ISR" functionality.  It shouldn't be difficult to port these over.  Thanks for raising this issue.

kobuki, 2020-09-19:
Thanks, that sounds somewhat reassuring. We won't have to look for alternatives.

joan2937, 2020-09-19:
Very little of pigpio uses the sysfs interface.  Most people will not be using sysfs when they use pigpio.  As far as I remember the only use of sysfs is in the gpioISR functions which can only be used if you link against the C library.  That means that Python users and pigpiod_if2 users will not be affected at all.

guymcswain, 2020-10-12:
According to www.kernel.org regarding "GPIO Sysfs Interface for Userspace":
> THIS ABI IS DEPRECATED, THE ABI DOCUMENTATION HAS BEEN MOVED TO Documentation/ABI/obsolete/sysfs-gpio AND NEW USERSPACE CONSUMERS ARE SUPPOSED TO USE THE CHARACTER DEVICE ABI. THIS OLD SYSFS ABI WILL NOT BE DEVELOPED (NO NEW FEATURES), IT WILL JUST BE MAINTAINED.

Since pigpio has no plans to add features using the sysfs interface and the deprecation warning implies on-going maintenance, we feel comfortable the five APIs in question will be fine as is.

closed 2020-10-12 by guymcswain

joan2937, 2020-10-12:
I would just like to comment that at the present time the /dev/gpiochip interface does not offer any compelling advantage over the deprecated sysfs GPIO interface.

If anything the improvements are marginal and mainly relate to time-stamping GPIO events at source (something pigpio already does).

kobuki, 2020-10-12:
> the only use of sysfs is in the gpioISR functions which can only be used if you link against the C library

If I plan to link against the C library (for SPI-related performance reasons), what is the recommended course of action? Will I be affected?

joan2937, 2020-10-12:
@kobuki No, you will not be affected.

The reason I mentioned the C library is that the only people who could be affected will be linking against the library directly.  Most people don't.

Of those that do link against the library the only people who could be affected are those that use the gpioISR functions.  Those functions are irrelevant to SPI.

As @guymcswain points out the people who currently use or want to use the gpioISR functions won't be affected either as there are no plans to remove the sysfs interface from the kernel at this stage.