joan2937/pigpio #388: Update version number of the Python interface.

innot, 2020-09-20:
Hi,

the version number of the python interface is currently 1.46. This number was set for the V75 release of pigpio.
While the new wave_create_and_pad() from V76 has already been added to the pigpio.py, the version number was not changed and therefore no new release of the Python interface was initiated.

As I am working on a small Python programm using pigpio and the new wave_create_and_pad() feature ( https://github.com/innot/AdvPiStepper ) and I would like to make the installation as easy as possible (via PyPi) it would be most helpful if the Python version number could be bumped up so that a new release of the python interface is initiated.

Regarding the version number: The Python version number seems somewhat arbitrary. How about getting the number in sync with the pigpio releases. So instead of going to 1.47, how about 1.76 to show that the Python interface has all the features of pigpio V76.

The version number is both in pigpio.py (line 334) and setup.py (line 6). 
Changing it should have no side effects other than marking the Python interface for an update on PyPi.org.

guymcswain, 2020-09-20:
Thanks for pointing this out.  

V78 is in progress, due to be released by the end of the month, so far without changes to the Python module.  V77 had changes to pigpio.py to correct some of the embedded documentation.  Missing a version change there was probably ok.  Following your recommendation,  going forward we should change the version of pigpio.py and setup.py to 1.78 in the next release to "sync up".  In a hypothetical V79, we would leave the Python version at 1.78 unless there was a change in that module.  Have I grasped your recommendation correctly?

Your suggestion seems reasonable to me but I'd like to hear others weigh in.

innot, 2020-09-21:
Yes, this is just what I had in mind. 
1.78 for the next release in the near future sounds good me.

guymcswain, 2020-09-29:
@joan2937 ,
Please upload the latest Python package to PyPI.

joan2937, 2020-09-29:
@guymcswain 

We don't maintain the PyPi package.  I think it is maintained by @bennuttall as part of his gpiozero work.

bennuttall, 2020-09-29:
Yep - will upload it later tonight. FYI I subscribe to releases and release to PyPI when I see them :+1:

bennuttall, 2020-09-29:
Uploaded: https://pypi.org/project/pigpio/

closed 2020-09-30 by guymcswain