joan2937/pigpio #389: Kotlin/Native sigHandler : unhandled signal 11, terminating

rkreienbuehl, 2020-09-28:
When I use pigpio with Kotlin/Native, I run into a `sigHandler : unhandled signal 11, terminating` error. This happens in the code below:

```
import kotlinx.cinterop.staticCFunction
import pigpio.*

const val GPIO_BUTTON = 6
var lastChange: UInt = 0u

fun main() {
    initGPIO()
    setupButton()

    while (true) {}
}

private fun initGPIO() {
    if (gpioInitialise() < 0) {
        println("GPIO Error initialising")
        return
    }
}

private fun setupButton() {
    val buttonPort = GPIO_BUTTON.toUInt()
    initPortWithMode(buttonPort, PI_INPUT)

    gpioSetAlertFunc(buttonPort, flankChangeDetected)
}

private fun initPortWithMode(port: UInt, mode: Int) {
    if (gpioSetMode(port, mode.toUInt()) < 0) {
        println("Could not set mode for GPIO$port")
        return
    }
}

val flankChangeDetected = staticCFunction<Int, Int, UInt, Unit> { gpio, level, tick ->
    val ticker = tick
    val pin = gpio
    val lvl = level

    val time = ticker - lastChange
    lastChange = ticker
    if ((time > 55u && time < 61u) || (time > 113u && time < 119u)) println(time)
}
```

The build.gradle file contains the folowing part:

```
kotlin {
    linuxArm32Hfp("omrr") {
        binaries {
            executable {
                entryPoint 'com.example.omrr.main'

                runTask?.args('')

                // libpigpio.so are compiled on raspberry pi and copied here
                linkerOpts("-Lsrc/lib/pigpio", "-lpigpio")

            }
        }

        compilations.main.cinterops {
            pigpio {
                includeDirs 'src/include/pigpio'
            }
        }
    }
}
```

The pigpio.def file contains the following line: `headers = pigpio.h`

To get started, I followed this talk:
[Bridge The Physical World: Kotlin Native on Raspberry Pi](https://speakerdeck.com/jinqian/bridge-the-physical-world-kotlin-native-on-raspberry-pi)

rkreienbuehl, 2020-09-28:
With some additional println I have figured out, that the error occurs on this line:
`lastChange = ticker`

jw3, 2020-09-28:
did you see the EMBEDDED_IN_VM flag?

rkreienbuehl, 2020-09-28:
If I add the `#define EMBEDDED_IN_VM` to the pigpio.h file, I get an additional output: "Speicherzugriffsfehler" (memory access failure) and the program quits.

guymcswain, 2020-09-28:
Please try a variant of the following in you gpio init function:
```
int cfg = gpioCfgGetInternals();
cfg |= PI_CFG_NOSIGHANDLER;
gpioCfgSetInternals(cfg);
int status = gpioInitialise();
```

rkreienbuehl, 2020-09-28:
I have tried this:
```
var cfg = gpioCfgGetInternals().toInt()
cfg = cfg or PI_CFG_NOSIGHANDLER
gpioCfgSetInternals(cfg.toUInt())
val status = gpioInitialise()
```

Status is 77, the error still occurs.

guymcswain, 2020-09-28:
I can only suggest you post an issue on the github repo referenced by your link.

willianwrm, 2020-10-07:
Having similar error in my project; using version 78 (tested all previous through 69) and getting random _Unhandled signal_ and _Segmentation fault._

I made a small one project that calls _gpioInitialise_ and then start a memory stress code (random buffer allocations), the issues does continue, so look like it's something at the _gpioInitialise_.

Using Raspberry Debian Buster (kernel 4.19.97), tested with Pi 3 and Pi4.

The same code runs smoothly at version 68 - but of course works only with Pi 3.

willianwrm, 2020-10-08:
> 
> 
> Please try a variant of the following in your gpio init function:
> 
> ```
> int cfg = gpioCfgGetInternals();
> cfg |= PI_CFG_NOSIGHANDLER;
> gpioCfgSetInternals(cfg);
> int status = gpioInitialise();
> ```

Using the PI_CFG_NOSIGHANDLER "solved" the problem 👍 
Basic I/O is working fine, didn't tested PWM yet.

guymcswain, 2020-10-12:
@rkreienbuehl 
> cfg = cfg or PI_CFG_NOSIGHANDLER

Is that a bitwise OR?

willianwrm, 2020-10-12:
> 
> 
> @rkreienbuehl
> 
> > cfg = cfg or PI_CFG_NOSIGHANDLER
> 
> Is that a bitwise OR?

Yes bitwise OR.
The idea is to get the current configuration and then add/flag the PI_CFG_NOSIGHANDLER.

guymcswain, 2020-10-13:
@willianwrm , et tu, Kotlin?

willianwrm, 2020-10-13:
> 
> 
> @willianwrm , et tu, Kotlin?

No

guymcswain, 2020-11-07:
@rkreienbuehl , I'm closing this issue with  the solution you posted on raspberrypi.org forum.

> After some trying I have found a solution, but don't know if it is the best way to do it.

> A global variable as an AtomicInt didn't work, but this way it works:
```
@SharedImmutable
val sharedData = SharedData()

class SharedData {
    var lastChange = AtomicInt(0)
}
```

> This way I can access the valua via val lastChange = sharedData.lastChange.value and sharedData.lastChange.compareAndSet(lastChange, tick.toInt())

closed 2020-11-07 by guymcswain