joan2937/pigpio #395: The distnace is less than 1 meters with anthena

roysG, 2020-10-12:
This library works great!
The data is transfer as expected, but the range is no so long, even with anthena.

Is there any configuration that may be help to make the distance longer?

joan2937, 2020-10-12:
I am not aware of any way to change the software to improve the range.

One metre seems poor.  I suspect either your receiver or your transmitter have a poor aerial.

I have only used the inexpensive 433 MHz modules (UK £2).  I tested at about 5 metres with a 16 cm length of wire soldered to the receiver and the transmitter as an aerial.

guymcswain, 2020-10-12:
You might also find better performance using the regenerative type receivers.  I have used these at 433mhz over 50 meters with several walls in the path.  Also, higher voltage on the transmitter will boost your power significantly.

closed 2020-10-12 by guymcswain