joan2937/pigpio #396: Pulse input arguments

mrzk, 2020-10-14:
Hi. I want to create a software serial signal with parity bit, but pigpio serial wave does not support it (I haven't seen the option on documentation). Due to ths, I decided to create my own serial signal but I can not understand documentation for Pulse object arguments.
gpio_on:= the GPIO to switch on at the start of the pulse.
gpio_off:= the GPIO to switch off at the start of the pulse.
delay:= the delay in microseconds before the next pulse.
These are the input arguments from which the 3rd one is understandable enough, but the other 2 I can not understand. Can you please clarify the other 2 arguments?

guymcswain, 2020-10-14:
>  I want to create a software serial signal with parity bit, but pigpio serial wave does not support it 

It can.  You just extend the size of the frame and do your own party generation and checking.

mrzk, 2020-10-14:
Thanks for your response. So if I extend my frame size, say 9 bits, it automatically resize the frame to 2 bytes (16 bit). What are the values for remaining 7 bits?

guymcswain, 2020-10-14:
They are "don't cares" and get truncated before serialization.

mrzk, 2020-10-14:
On receive you mean on receiver side? If so I don't have access to the receiver, it's a device which I can't alter its code

guymcswain, 2020-10-14:
I'm closing this issue: To summarize the software serial APIs are:
`gpioWaveAddSerial` (et al) for transmit
`gpioSerialRead*` for receive
or their equivalent for socket interface.

closed 2020-10-14 by guymcswain