joan2937/pigpio #399: pigpio conflicts with audio playback

smdjeff, 2020-10-25:
Audio playback still seems to I2C/PCM work, but seems to be playing at half speed.

smdjeff, 2020-10-25:
A workaround for the audio playback problem seems to be calling gpioCfgClock() with the PWM setting. Why is PCM the default given it will conflict with audio?

The documentation for gpioCfgClock() isn't clear. The description seemed to suggest it was used to configure a sample rate for both the PWM and PCM peripherals. However, its main purpose is to select which of the two peripherals are to be used by the gpio library while also configuring the sample rate for that peripheral.
http://abyz.me.uk/rpi/pigpio/cif.html#gpioCfgClock

joan2937, 2020-10-25:
Start the pigpio daemon with the -t0 option to use the PWM chip for timing.

http://abyz.me.uk/rpi/pigpio/pigpiod.html

joan2937, 2020-10-25:
http://abyz.me.uk/rpi/pigpio/faq.html#Sound_isnt_working

closed 2020-10-25 by smdjeff