joan2937/pigpio #400: PI_CMD_EVT  is unused  in pigpiod_if2.c

satoren, 2020-10-27:
Is this `PI_CMD_EVT` correct?
https://github.com/joan2937/pigpio/blob/cc17196c7586c3625ac8adf8f34838d320ffeda6/pigpiod_if2.c#L2125

joan2937, 2020-10-28:
Probably not.  The equivalent command in pigpio.py uses EVT not EVM.

If you change the code to PI_CMD_EVT does it do what you expect?

satoren, 2020-10-28:
I haven't tried it. I'm working on a compatible library in Node.js. I noticed this when I was referring to the C code at the time.

guymcswain, 2020-10-28:
> I'm working on a compatible library in Node.js.

Have a look at [pigpio-client](https://github.com/guymcswain/pigpio-client).  It may save you some time.  Most APIs have been implemented, including this one, and it's fairly easy to add others.

satoren, 2020-10-28:
I like type(script).

guymcswain, 2021-03-02:
Fixed in v79

closed 2021-03-02 by guymcswain