joan2937/pigpio #401: Wiegand example code has decoding error.

jacoka, 2020-10-29:
Hi,
 I use your wiegand example code for a project that use soyal nfc readers to read mifare nfc keys.
Seemed to be everything works well till i bought a mifare dekstop key reader, and i saw with sorrowfull the serial number of keys were not same during read keys.
I made an html application (https://zsidi.hu/nfc) this application can be used to read serial number of mifare key with.
I tried again compare to read serial numbers with this application but the result was same.

Reading error was generated by too wide bit length (your original code try to set serial code on 34 bits, but int32 have 32 bits only).

I have modified your code and works perfectly.
`	if(level == 0) {
		if(!wieg->in_code) {
			wieg->bits = 1;
			wieg->num = 0;
			wieg->in_code = 1;
			wieg->code_timeout = 0;

			gpioSetWatchdog(wieg->mygpio_0, wieg->mytimeout);
			gpioSetWatchdog(wieg->mygpio_1, wieg->mytimeout);
		}
		else {
			if(wieg->bits++ <= 32) {
				wieg->num <<= 1;
			}
		}

		if(gpio == wieg->mygpio_0) {
			wieg->code_timeout &= 2;
		}

		if(gpio == wieg->mygpio_1) {
			wieg->code_timeout &= 1;
			if(wieg->bits <= 33)
				wieg->num |= 1;
		}
	}`

Sorry for may bad english :)
Thank you for your code!

guymcswain, 2020-11-06:
I would be happy to accept a pull request or a patch with the specific code change(s).

guymcswain, 2020-12-19:
> I have modified your code and works perfectly.

@jacoka , do you care to share your modification?  If not I'll be closing this issue.  Thanks in advance.

closed 2021-03-20 by guymcswain