joan2937/pigpio #402: Cannot handle multiple I2C devices on RPi 4B

poirierlouis, 2020-11-04:
**Context**
Raspberry Pi : v4 Model B
pigpio C library : v78
Use multiple MCP23017 devices with I2C protocol.

**Problem**
When initializing multiple devices, first device with handle 0 (returned by `i2cOpen`) can be used without problem. I can write with `i2cWriteByteData`.
But using another device with handle 1, I can't write anymore as the library returns a PI_BAD_HANDLE error code when using `i2cWriteByteData`. While the call to `i2cOpen` for second device, returned a valid handle 1.

**Example**
``` c
#include "pigpio.h"

#define MCP23017_ADDRESS 0x20
#define MCP23017_IODIRA 0x00
#define MCP23017_IODIRB 0x01

#define I2C_BUS 1
#define I2C_FLAGS 0

int mcp[2];

if (gpioInitialise() == PI_INIT_FAILED) { return; }

mcp[0] = i2cOpen(I2C_BUS, MCP23017_ADDRESS | 0b000, I2C_FLAGS);
i2cWriteByteData(mcp[0], MCP23017_IODIRA, 0xFF);
i2cWriteByteData(mcp[0], MCP23017_IODIRB, 0xFF);

mcp[1] = i2cOpen(I2C_BUS, MCP23017_ADDRESS | 0b001, I2C_FLAGS);
i2cWriteByteData(mcp[1], MCP23017_IODIRA, 0xFF); // return PI_BAD_HANDLE
i2cWriteByteData(mcp[1], MCP23017_IODIRB, 0xFF);
```

guymcswain, 2020-11-06:
Thanks for finding this.  It'll be in the next release before the year end and in the develop branch sooner.

guymcswain, 2021-03-02:
Fixed in v79

closed 2021-03-02 by guymcswain