joan2937/pigpio #408: [pigpiod_if2] always returning 0 in get_PWM_dutycycle does not make sense

hulloanson, 2020-11-23:
As far as I understand, this function is supposed to get current dutycycle of a gpio pin, which the equivalent command would be `pigs gdc 18` if I wanted to get dutycycle of pin 18.

Yet as mentioned in documentation and in actual runs, it always returns 0. What is it supposed to accomplish? How do I get pwm dutycycle with the pigpiod_if2 api?

[image]

joan2937, 2020-11-24:
Given that it works properly for me I suggest you add example code which does not work for you.

hulloanson, 2020-12-07:
I am sorry. There was a problem with outputting the reading on my side. It now returns the correct number :)

I still suggest we modify the documentation to make it clear though.

hulloanson, 2020-12-07:
Pull request: https://github.com/joan2937/pigpio/pull/413

guymcswain, 2021-03-02:
Fixed in v79

closed 2021-03-02 by guymcswain