joan2937/pigpio #409: PigpioSocket v2.4 gives (sometimes) ClassCastException

cvdenzen, 2020-11-24:
Every now and then (maybe when the pigpiod server is not running?) a call to PigpioSocket.i2cReadI2CBlockData throws an exception:
```
Caused by: java.lang.ClassCastException: class uk.pigpioj.PigpioSocket$ResponseMessage cannot be cast to class uk.pigpioj.PigpioSocket$ByteArrayResponseMessage (uk.pigpioj.PigpioSocket$ResponseMessage and uk.pigpioj.PigpioSocket$ByteArrayResponseMessage are in unnamed module of loader org.apache.felix.framework.BundleWiringImpl$BundleClassLoader @ed3eb6)
        at uk.pigpioj.PigpioSocket.i2cReadI2CBlockData(PigpioSocket.java:738) ~[!/:2.4]
        at nl.vandenzen.iot.beans.BME280Driver.read(BME280Driver.java:268) ~[!/:?]
        at nl.vandenzen.iot.beans.BME280Driver.getSensorValues(BME280Driver.java:386) ~[!/:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.apache.camel.support.ObjectHelper.invokeMethodSafe(ObjectHelper.java:352) ~[!/:3.6.0]
        at org.apache.camel.component.bean.MethodInfo.invoke(MethodInfo.java:474) ~[!/:3.6.0]
        at org.apache.camel.component.bean.MethodInfo$1.doProceed(MethodInfo.java:296) ~[!/:3.6.0]
        at org.apache.camel.component.bean.MethodInfo$1.proceed(MethodInfo.java:266) ~[!/:3.6.0]
        at org.apache.camel.component.bean.AbstractBeanProcessor.process(AbstractBeanProcessor.java:145) ~[!/:3.6.0]
        at org.apache.camel.component.bean.BeanProcessor.process(BeanProcessor.java:54) ~[!/:3.6.0]
        at org.apache.camel.processor.errorhandler.RedeliveryErrorHandler$SimpleTask.run(RedeliveryErrorHandler.java:404) ~[!/:3.6.0]
        at org.apache.camel.impl.engine.DefaultReactiveExecutor$Worker.schedule(DefaultReactiveExecutor.java:148) ~[!/:3.6.0]
        at org.apache.camel.impl.engine.DefaultReactiveExecutor.scheduleMain(DefaultReactiveExecutor.java:60) ~[!/:3.6.0]
        at org.apache.camel.processor.Pipeline.process(Pipeline.java:147) ~[!/:3.6.0]
        at org.apache.camel.processor.CamelInternalProcessor.process(CamelInternalProcessor.java:288) ~[!/:3.6.0]
        at org.apache.camel.impl.engine.DefaultAsyncProcessorAwaitManager.process(DefaultAsyncProcessorAwaitManager.java:83) ~[!/:3.6.0]
        at org.apache.camel.support.AsyncProcessorSupport.process(AsyncProcessorSupport.java:41) ~[!/:3.6.0]
        at org.apache.camel.component.quartz.CamelJob.execute(CamelJob.java:62) ~[!/:3.6.0]
        ... 2 more
```

joan2937, 2020-11-24:
That does not seem to be part of pigpio.  It seems like it is a third-party library.  You need to raise the issue with the author of the third-party library.

cvdenzen, 2020-11-25:
Yes, you are right! I will raise the issue with the third-party. Thank you for your quick reply!

closed 2020-11-26 by guymcswain