joan2937/pigpio #410: gpioTerminate() hangs preventing program close using buildroot package

stevecool2020, 2020-11-24:
Using buildroot (git hash 2ef3e19b7b026ad1f6839092428419011ba5d0f4) and pulling in pigpio package for linux kernel build  Linux 4.19.97-v8 aarch64 on Raspberry Pi 4B

I've setup a small program that tests the pigpio library. My program successfully initializes the library with `gpioInitialise()` but hangs indefinitely on `gpioTerminate()`.

`pigpiod -v` reveals the following: 7301

I'm not sure what version this is supposed to be.

Any ideas? Do I need an more updated buildroot package for pigpio?

guymcswain, 2020-11-25:
The current release is 78.  A four digit version is use to denote a pre-release usually on the develop branch.  This one, 7301, may have escaped into the wild.

guymcswain, 2020-12-19:
@stevecool2020 , was the problem resolved after upgrading the most recent version?  If so, I'd like to close this issue.

closed 2021-03-19 by guymcswain

alpo, 2021-05-04:
gpioTerminate() still hangs in pipgpio v78 built under Buildroot 2021.02.