joan2937/pigpio #411: gpioStartThread

smdjeff, 2020-11-28:
I think the documentation for gpioStartThread is missing a return argument in the example for is 
void *myfunc(void *arg) { }

warning: control reaches end of non-void function
      [-Wreturn-type]

http://abyz.me.uk/rpi/pigpio/cif.html#gpioStartThread

guymcswain, 2020-11-28:
I don't see a problem with the example.

joan2937, 2020-11-28:
The warning is for main() which as declared should return an int.  Perhaps we should add a **return 0;** after the **gpioTerminate();**.

smdjeff, 2020-11-28:
myfunc() should return a pointer to.... Something apparently. It's a void* return type. What does it return?

joan2937, 2020-11-28:
It should be **void myfunc** not **void *myfunc**.  It looks like the prototype is wrong.  I'm not going to change it.  Just accept it is wrong and ignore it.

joan2937, 2020-11-28:
Not my day.  The prototype and code are correct.  Nothing is returned as the thread does not terminate.  It is a while forever loop.

The following is a better example.

```c
#include <stdio.h>
#include <unistd.h>

#include <pigpio.h>

void *myfunc(void *arg)
{
   while (1)
   {
      printf("%s\n", (char *)arg);
      sleep(1);
   }
}

int main(int argc, char *argv[])
{
   pthread_t *p1, *p2, *p3;

   if (gpioInitialise() < 0) return 1;

   p1 = gpioStartThread(myfunc, "thread 1"); sleep(3);

   p2 = gpioStartThread(myfunc, "thread 2"); sleep(3);

   p3 = gpioStartThread(myfunc, "thread 3"); sleep(3);

   gpioStopThread(p3); sleep(3);

   gpioStopThread(p2); sleep(3);

   gpioStopThread(p1); sleep(3);

   gpioTerminate();

   return 0;
}
```

smdjeff, 2020-11-28:
gpioThreadFunc_t returns a pointer. gpioStartThread() calls pthread_create so the return from gpioThreadFuncs are passed to pthread_exit(). A return of NULL here would remove the warning and improve documentation.