joan2937/pigpio #412: pigpiod not working on Raspberry 400

ilguargua, 2020-11-30:
`pi@raspberrypi:~ $ cat /etc/issue
Raspbian GNU/Linux 10 \n \l

pi@raspberrypi:~ $ pigpiod -v
71
pi@raspberrypi:~ $ sudo pigpiod 
2020-11-30 09:01:41 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, this system does not appear to be a raspberry pi. |
|aborting.                                                |
+---------------------------------------------------------+

Can't initialise pigpio library
pi@raspberrypi:~ $ 
`
The system is regulary update and upgraded, I don't know what else I could try.

Thank you,

Ale.

joan2937, 2020-11-30:
Could you let us know the result of the following command.

**od -x /proc/device-tree/system/linux,revision**

On a PiB+
```
0000000 0000 1000
0000004
```
On a Pi4B
```
0000000 a000 1131
0000004
```

joan2937, 2020-11-30:
For information a Pi 400 returns
```
0000000 c000 3031
0000004
```
https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=292890#p1770396

guymcswain, 2020-11-30:
So we need to update the 'initialise' function.  I can patch this in develop branch for testing.

@ilguargua , how did you install to get pigpio v71?

joan2937, 2020-11-30:
@guymcswain I don't understand why 00c0 3130 (from c000 3031) wouldn't be detected as a BCM2711 and work.  Am I being silly?

guymcswain, 2020-11-30:
@joan2937 , no between old, new revision codes and LE vs BE it gets confusing.  I'm starting to dig through it again to understand it myself.

guymcswain, 2020-11-30:
@ilguargua , please show the output from this command:
`grep Revision /proc/cpuinfo`

ilguargua, 2020-11-30:
Hi guys,
thanks for all propmt replays.
I don't have my 400 with me right now (I left it at work), so I'll let you know tomorrow morning (21:22 pm now here)

Thank you again for support,

Cheers, Ale.

ilguargua, 2020-11-30:
@guymcswain : pigpio was installed with the Raspian (full version) that came with pi, after been regulary updated.

Ciao, Ale.

ilguargua, 2020-12-01:
@guymcswain : 
`pi@raspberrypi:~ $ grep Revision /proc/cpuinfo
Revision	: c03130
pi@raspberrypi:~ $ sudo pigpiod -c4 -g 2>&1 | grep gpioHardwareRevision
2020-12-01 07:41:55 gpioHardwareRevision: 
pi@raspberrypi:~ $ 
`
Thank you for your support, let me know if you need something else.

Ciao, Ale.

guymcswain, 2020-12-01:
Scratching my head ... I was expecting several output lines from the second grep command.

The output from /proc/cpuinfo looks good, ie, pigpio should have detected this as a valid raspberrypi.

guymcswain, 2020-12-01:
@ilguargua ,
The logic to detect hardware revision was updated since v74.  Please update your pigpio version from v71 to the latest v78 as follows:
```bash
wget https://github.com/joan2937/pigpio/archive/master.zip
unzip master.zip
cd pigpio-master
make
sudo make install
```
If the Python part of the install fails it may be because you need the setup tools.
```bash
sudo apt install python-setuptools python3-setuptools
```

ilguargua, 2020-12-01:
@guymcswain ,
everything works!!
I first launch : sudo apt-get remove pigpio pigpiod pigpio-tools , followed by your instruction.
We should strongly suggest to the Raspian people to update the repository version, since is proved not working on Raspy 400.

Thanks a lot to eveyone, should now this issue be closed?

Ciao, Ale.

guymcswain, 2020-12-01:
Thanks for your help.

closed 2020-12-01 by guymcswain