joan2937/pigpio #414: How can the spi clock frequency be set?

OzyOzk, 2020-12-07:
I'm running the code below to benchmark an spi microphone (PMOD MIC3). This code is based on an example I found on Stack Overflow. The microphone module I'm using uses an ADC that is rated to work with clock-speeds between 10kHz - 20MHz. At the moment I'm not sure what clock speed pigpio uses. I can set the baud rate but this doesn't tell me what the clock frequency is.

#include <stdio.h>
#include <stdlib.h>
#include <pigpio.h>
#include <unistd.h>

#define BAUD_RATE 500000
```
double pmod_mic3(double* in_array, int size){
  if (gpioInitialise() < 0) return 1;
  ts.tv_nsec = 2300;
  ts.tv_sec = 0;
  double start, delta;
  double analog = 0;
  int h = spiOpen(0, BAUD_RATE, 0);
  if (h < 0) return 1;
  
  /*bits to send*/
  unsigned char snd[2] = {0, 0};
  
  /*buffer*/
  unsigned char buf[2] = {0, 0};
  
  start = time_time();

  for (int i=0; i<size; i++){
    spiXfer(h, snd, buf, 2);
    analog = buf[0] << 8 | buf[1];
    in_array[i] = analog;
  }
  
  delta = time_time() - start;
  spiClose(h);
  return delta;
}
```
With a BAUD_RATE set to 500Khz I get around 57K samples a second. I'm using an RPI4 (4GB) if that is helpful.

guymcswain, 2020-12-07:
> At the moment I'm not sure what clock speed pigpio uses.

I don't think this question is relevant.  You need to research the documentation of the module you are working with.  Presumably it has its own clock or perhaps it is derived from SPI (which is the baud rate).  You may want to post your question on the raspberrypi.org forum to see if other users of the microphone module can help you.

OzyOzk, 2020-12-08:
@guymcswain the microphone relies on SCLK coming from the master, it doesn't have a clock of it's own. It's just an electret mic connected to a TI ADCS747x ADC. The ADC is rated to work from 10Khz to 20Mhz. 

What I'm not sure about is the second argument of the spiOpen function. Is this baud rate or clock speed?

guymcswain, 2020-12-09:
It is the same: SCK frequency = baudrate.

closed 2020-12-18 by guymcswain