joan2937/pigpio #417: Map to SPI/BCS SLAVE peripheral SFRs needs to be parsed to “volatile uint32_t *” (pigpio.c, line 7450)

smeerpate, 2020-12-13:


guymcswain, 2020-12-19:
I don't understand.  Is there a problem you are experiencing with the library?

smeerpate, 2020-12-20:
Hi,

I was using the RPi BSC slave as an i2c slave and left it run for over a day and a half or so.

I am periodically disabling the peripheral by writing 0 to the I2C bit in the CR register. I’m doing this because i need the i2c slave to NACK a read request from my i2c master and I’m doing so using the control flags in my bsc_xfer_t instance.

After some period, the peripheral wasn’t ACK-ing the i2c master’s read request even when it was supposed to do so. The i2c bit in my bsc_xfer_t instance was effectively set.

Still in this situation, I mmaped the peripheral’s registers using some other code and noticed that the i2c bit there was cleared.

I’m very sorry to have posted this issue and to have bothered you with this. But I’ve changed this in my local fork and didn’t come across this issue any longer. Does this makes sense to you?

Best regards,
Frederic

guymcswain, 2020-12-20:
No I'm not bothered, I just needed more information so thanks for providing it.

I think I follow what your are doing with the peripheral but sharing a code snippet would be very helpful.

I'm scratching my head over how mapping to volatile resolves the issue.  If there was some kind of coherency issue due to the way the peripherals are mapped then many parts of the library would be falling over.

guymcswain, 2020-12-20:
I think I found your code here?
https://github.com/smeerpate/SACI2cIotSlave/blob/main/SACRPiIotSlave.c

guymcswain, 2020-12-20:
Maybe I'm missing it but I don't see where in your code **all** the flags are initialized.

smeerpate, 2020-12-20:
Hi Guy,
It was my understanding that this was done in the getControlBits function in SACRPiIotSlave.c...

> I think I found your code here?
> https://github.com/smeerpate/SACI2cIotSlave/blob/main/SACRPiIotSlave.c
Correct.

guymcswain, 2020-12-20:
Yes, but I only see a subset of the bits getting set.  Also, not sure if they are cleared when a 0 gets or'ed for that position.

guymcswain, 2020-12-20:
To verify, this same code failed before you edited the single line in pigpio and is now working?

smeerpate, 2020-12-21:
Hi Guy,
It seems so on my UsingGpioLib-branch.
I will verify that to be 100% sure and I will get back to you on this later this week.
Thanks for your help!
Frederic

smeerpate, 2020-12-24:
Hi Guy,
I have checked this again, and as you suspected, the volatile statement doesn’t change a thing. I still don’t understand why I’m reading two different values in the control register.
My program here outputs this:
[image]
This is what I read back when I’m using another program just after I read back 0x305 from CR using mmap:
[image]
I finally moved over to using the serial port with termios instead because I ran out of time...
Best regards,
Frederic

guymcswain, 2021-01-03:
Hi Frederic,

I'm going to write a test that will try to simulate the conditions in your environment.  To that end I have a few questions:
1. Which RPi model are you using?
2. Is the I2C master external and if so what is it?
3. OS and pigpio versions?

Thanks,
Guy

smeerpate, 2021-01-10:
Hi Guy,

1. cat /proc/device-tree/model:
> Raspberry Pi 3 Model B Rev 1.2

2. External I2c master:
> Arduino micro (P/N A000053)

3a. cat /etc/os-release:
> PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
> NAME="Raspbian` GNU/Linux"
> VERSION_ID="9"
> VERSION="9 (stretch)"
> ID=raspbian
> ID_LIKE=debian

3b. pigpio version:
> pigpio version 78

Does all above info makes sense?

I admire your commitment!

Best regards,
Frederic