joan2937/pigpio #421: How difficult to add callback functionality for one pin in Bank 2?

nanok66, 2021-01-07:
Hi,  I see that your library only supports Bank 1 GPIO callbacks, but I was just curious if I only wanted to add functionality for one or two currently unsupported pins would it be terribly hard?  

I'm using the python module with a Pi Compute 3+ trying to make callbacks work for pin 35. 

It's probably way over my head but if it's not too difficult could you share any examples of the code changes needed to do so? 

I appreciate any advice you can give, thanks!!

guymcswain, 2021-01-08:
Not difficult at all.  You'll need to fork the library and edit the pigpio.c file and then follow the instructions on how to make and install the library - in this case your custom version.  (If any of that sounds difficult to you, that's a red flag.)

I haven't looked specifically at the callback API but I know that numerous checks are made to verify that the gpio you specify is a valid 'user gpio'.  I think it should be as simple as to bypass that check in the callback API.

joan2937, 2021-01-08:
Use the ISR functions as they work on all GPIO and will do what you want.

ISR callbacks have lower latency but a less reliable timestamp. Work on all GPIO.

Alert callbacks have higher latency but a more reliable timestamp. ONLY work on GPIO031.

 It would be a major change to extend alter callbacks to the other GPIO.

nanok66, 2021-01-10:
Great, will follow both your advice to create this fork!   Thanks for the
quick responses!

guymcswain, 2021-01-11:
> It would be a major change to extend alert callbacks to the other GPIO.

You should heed the above advice from the author of the pigpio library.  Try to make your application work with ISR callbacks before you take the plunge into hacking the 'alert' type callbacks.

closed 2021-03-19 by guymcswain