joan2937/pigpio #423: Question about the python example "433MHz Keyfob RX/TX"

GF-Huang, 2021-01-11:
In the `rx._cbf(self, g, l, t)`:

```py
def _cbf(self, g, l, t):
   """
   Accumulates the code from pairs of short/long pulses.
   The code end is assumed when an edge greater than 5 ms
   is detected.
   """
   edge_len = pigpio.tickDiff(self._last_edge_tick, t)
   self._last_edge_tick = t

   if edge_len > 5000: # 5000 us, 5 ms.

      if self._in_code:
         if self.min_bits <= self._bits <= self.max_bits:
            self._lbits = self._bits
            self._lcode = self._code
            self._lgap = self._gap
            self._lt0 = int(self._t0/self._bits)
            self._lt1 = int(self._t1/self._bits)
            self._ready = True
            if self.cb is not None:
               self.cb(self._lcode, self._lbits,
                        self._lgap, self._lt0, self._lt1)

      self._in_code = True
      self._gap = edge_len
      self._edge = 0
      self._bits = 0
      self._code = 0

   elif self._in_code:

      if self._edge == 0:
         self._e0 = edge_len
      elif self._edge == 1:
         self._calibrate(self._e0, edge_len)

      if self._edge % 2: # Odd edge.

         bit = self._test_bit(self._even_edge_len, edge_len)
         self._code = self._code << 1
         if bit == 1:
            self._code += 1
         elif bit != 0:
            self._in_code = False

      else: # Even edge.

         self._even_edge_len = edge_len

      self._edge += 1
```

> The code end is assumed when an edge greater than 5 ms is detected.

Because there is always noise, I don't quite understand when a 5000us pulse appears before you press the button on the remote control. So does it means the signal of your first time to press button will be ignored?

guymcswain, 2021-01-11:
Typically these inexpensive OOK transmitters generate 3 or more code sequences sequentially - as seen in your attached image.  So it shouldn't matter that the code is looking for sync on the trailing edge.  You could alter the code to look for sync on the leading edge if it is important to your application.

GF-Huang, 2021-01-11:
Got it, thanks.

closed 2021-01-11 by GF-Huang