joan2937/pigpio #424: Running PIGPIO on the new CM4

JesperNaarttijarvi, 2021-01-14:
Is it an easy feature to add? Right now we get the error "Sorry, this system does not appear to be a raspberry pi. aborting." when running it on the new compute module 4.

JesperNaarttijarvi, 2021-01-14:
My fault. Cloned the repo instead of apt-get. Now it starts, did not pass all the tests though.

closed 2021-01-14 by JesperNaarttijarvi

hulloanson, 2021-02-25:
For reference to other facing the same problem, cloning and compiling from v1.78 source works.

I tried installing 1.71-rpt from apt. Prompted the same `Sorry, this system does not appear to be a raspberry pi. aborting.` message.

hulloanson, 2021-02-26:
Another note: `apt-get install python3-distutils` if `make install` threw `ModuleNotFoundError: No module named 'distutils.core'` when installing pigpio to python3