joan2937/pigpio #425: pigpiod running on a ESP32 ??

louisvangeldrop, 2021-01-17:
I am using the pigpiod running on a Raspberry Pi Zero W. Nice, but expensive compared to a ESP32. The pigpiod is an excellent tool and could  be used on more devices than a Raspberry Pi. I hope you like my suggestion?

joan2937, 2021-01-18:
pigpio's most unique features are the DMA sampling and time-stamping of GPIO and the DMA generation of accurately timed GPIO level changes.  These functions are implemented on the Broadcom SoC only available in Raspberry Pis.

If you can port those features to another device you can port pigpio.

Without those features pigpio is pretty much a standard C library.

louisvangeldrop, 2021-01-18:
I will have a look at the moddable javascript engine. It fully supports the ESP32 and if performance is an issue I always can use a C-routine.

closed 2021-01-18 by louisvangeldrop