joan2937/pigpio #426: What happens when a GPIO line changes while handling a previous change (gpioSetISRFunc)?

edahlseng, 2021-01-18:
Thank you for this awesome library!

I have a clarification question about the behavior of the callbacks resulting from `gpioSetISRFunc` – while handling a GPIO line change with a `gpioSetISRFunc` handler, what happens if there's another change in the GPIO line? I see that this library uses the sysfs GPIO interface, but I can't find any documentation on whether that interface buffers changes or not.

This is particularly relevant when working with e.g. the [HX711](https://cdn.sparkfun.com/datasheets/Sensors/ForceFlex/hx711_english.pdf), which uses the same GPIO pin for indicating when data is available, as well as for transmitting that data. I'm hoping that the "interrupts" through this interface are _not_ buffered, and while some early testing seems to suggest that's the case, I can't find any definitive documentation on it.

Thank you for the help.

guymcswain, 2021-01-19:
> I see that this library uses the sysfs GPIO interface, but I can't find any documentation on whether that interface buffers changes or not.

Assume it doesn't.  You can use the initial falling edge of Dout to trigger a data available interrupt but then you should use the callback API to get the data level on each SCK.  Those callbacks are buffered.  HTH

edahlseng, 2021-02-24:
Thanks for your help @guymcswain!

> but then you should use the callback API

Which callback API are you referring to? Are you suggesting an approach other than using `gpioRead`?

What I'm currently doing is loosely:

```c
void getLoad(int gpio, int level, uint32_t tick) {
    static int32_t nextBit = 0;
    static int32_t load = 0;
    load = 0;

    if (level == 0) {
        for (int i = 24; i > 0; i--) {
            gpioWrite(CLOCK_GPIO_PIN, 1);
            gpioWrite(CLOCK_GPIO_PIN, 0);

            nextBit = gpioRead(DATA_GPIO_PIN);

            load <<= 1;
            load |= nextBit;
        }
    }

    ...
}

int main(int argc, char *argv[]) {
    ...

    gpioSetISRFunc(DATA_GPIO_PIN, FALLING_EDGE, 1000, getLoad);

    ...
}
```

guymcswain, 2021-02-24:
The specific callback API depends on how you access the library.  I see now that you are using the C interface, so the API I would use is called `gpioSetGetSamplesFunc`.  This will give you all the events on Clock and Data so you can sample the data on the clock falling edge (also, timestamps are available so that you can detect pulse widths and so forth).

However, looking at your code above, it seems to me that if you turn off the ISR callbacks, so that the ISR is not re-entrant, then do your clocking and data sampling, then re-enable ISR on exit, it may work.  To cancel the ISR you pass it a null for function.  The caveat to this is that the GPIO gets unexported then re-exported by sysfs, but pigpio can still write/read the GPIO while it is unexported.

@joan2937 , do you have another suggestion?

edahlseng, 2021-03-04:
Thank you for your input @guymcswain. I'm curious on the amount of overhead associated with turning the ISR off and on, but that's easy enough for me to test! Interestingly enough, I appear to be seeing the desired behavior without explicitly turning the ISR off and then back on – I'm just not sure if I'm getting "lucky" or if that is how the ISRs are designed.

joan2937, 2021-03-05:
I think Linux queues up one more interrupt for a GPIO while the current one is being processed.  You shouldn't need to switch ISRs on and off unless you have an odd requirement.

pigpio does nothing special with ISRs so your best bet is a search for Linux interrupt handling if you need to find out what is going on.

guymcswain, 2021-03-19:
Closing, don't think there is an issue here to follow up on.

closed 2021-03-19 by guymcswain