joan2937/pigpio #429: Confusing versioning of the project

pemensik, 2021-01-31:
I am just trying to prepare Fedora binary package, because so far only python3-pigpio package exists, but no daemon.

However, I found multiple differences scattered across the git repository. One is in setup.py, but it is not related to recent tag. Another in pigpioConfigVersion.cmake. It is not clear to me, whether official project version is 0.78, 1.78 or just 78.

Is the version just 78, as hinted by git tag? Could it be autogenerated from single source and to the same value after release?

joan2937, 2021-02-01:
I hear your pain.  I screwed up versioning big time.  I would treat the whole archive as a suite and consider it all version 78 (as of now).  In practice different versions of components are interoperable, they will just report errors if you try to use a later feature.

guymcswain, 2021-02-13:
In responding to an issue/request I 'synchronized' the minor number of the python module version with the core pigpio.c, ie 78.  IIRC this was done because of confusion about with versions supported 'padded' waves.  Going forward I was planning to keep this number synchronized but I'm open to other suggestions.

closed 2021-03-19 by guymcswain