joan2937/pigpio #430: Timing-problems at waveforms and wavechains

RolfKausH, 2021-02-03:
I use your PIGPIO library on a „Raspbian Pi 3 Model B V 1.2“ for sending Infrared signals. That did not work.

Using your tool „irrp.py“ I can read the signals sent from my remote-control. But sending this to my TV did not work because of wrong timing (??)

I recorded the wave while sending it – both with „irrp.py“. This is the result („PWR“ recorded from the remote-control, „PWR1“ recorded while sending)
{"PWR": [900, 2299, 459, 1633, 459, 938, 459, 938, 459, 1633, 459, 1633, 459, 1633, 459, 938, 459, 938, 459, 938, 459, 1633, 459, 938, 459, 1633, 459, 938, 459, 1633, 459, 1633, 459, 938, 459, 1633, 459, 938, 459, 1633, 459],
"PWR1": [1844, 4563, 967, 3249, 967, 1773, 967, 1773, 967, 3249, 967, 3249, 967, 3249, 967, 1773, 967, 1773, 967, 1773, 967, 3249, 967, 1773, 967, 3249, 967, 1773, 967, 3249, 967, 3249, 967, 1773, 967, 3249, 967, 1773, 967, 3249, 967]}
  
I startet the test-scripts:
x_pigpio.py
Connected to pigpio daemon.
Testing pigpio Python module 1.78
Python 2.7.16 (default, Oct 10 2019, 22:02:15)  [GCC 8.3.0] pigpio version 78.
Hardware revision 10625154.

reported errors:
TEST  5.4  FAILED got 25 (callback: 50) 
TEST  5.10 FAILED got 1743 (wave tx busy, callback: 1702) 
TEST  5.11 FAILED got 0 (wave tx busy, serial read: 1) 
TEST  5.26 FAILED got 25 (callback: 50) 
TEST 13.7 FAILED got 2688 (glitch filter, wave chain, tally: 1792) 
TEST 13.8  FAILED got 1792 (glitch filter, wave chain, tally: 896) 
TEST 13.9  FAILED got 1792 (glitch filter, wave chain, tally: 0) 
TEST 13.10 FAILED got 748 (noise filter, wave chain, tally: 1500) 
TEST 13.12 FAILED got 13 (noise filter, wave chain, tally: 0)
  
I created a work-around, because it seams that sending takes nearly twice the time:
in "irrp.py" I created a new line between 463 (ci = code[i]) and 464 (if i and 1:)
"ci = int(ci/2)"
This pulse-tuning works good for me. Maybe there is a better way, but it works.

guymcswain, 2021-02-13:
Something is wrong in your system configuration.  You should be able to run a clean (100% pass) of the python tests.

> reported errors:
TEST 5.4 FAILED got 25 (callback: 50)
TEST 5.10 FAILED got 1743 (wave tx busy, callback: 1702)
TEST 5.11 FAILED got 0 (wave tx busy, serial read: 1)
TEST 5.26 FAILED got 25 (callback: 50)
TEST 13.7 FAILED got 2688 (glitch filter, wave chain, tally: 1792)
TEST 13.8 FAILED got 1792 (glitch filter, wave chain, tally: 896)
TEST 13.9 FAILED got 1792 (glitch filter, wave chain, tally: 0)
TEST 13.10 FAILED got 748 (noise filter, wave chain, tally: 1500)
TEST 13.12 FAILED got 13 (noise filter, wave chain, tally: 0)

guymcswain, 2021-02-13:
You may want to experiment with setting `core_freq=250` in `/boot/config.txt`.

dgalland, 2021-02-14:
I have a similar problem with pigpio controlling a stepper motor.
The application was working correctly but recently, probably after rpi-update, the motor is running half as fast.
If I multiply the frequency of the wave by two the speed is good as before !
Kernel version:
Linux raspberrypi 5.10.14-v7+ #1401 SMP Mon Feb 8 14:23:24 GMT 2021 armv7l GNU/Linux

guymcswain, 2021-02-14:
Please verify that you have not enabled audio inadvertently - connecting hdmi to monitor,  enabling sound in config.txt.

Try running the daemon using the alternate clock source, PWM: `sudo pigpio -t 0`

dgalland, 2021-02-15:
- No audio or HDMI
- With the alternate source normal behavior and normal speed for the motor.
This factor 2 is intriguing, there have probably been changes in the firmware and/or kernel (with rpi-update I get the version under development)
Not blocking for me but tell me if I need to do more tests.
Thanks !

@RolfKausH 
Did you update you Pi recently ?

guymcswain, 2021-02-15:
> With the alternate (clock) source normal behavior and normal speed for the motor.

Something is now altering the PCM peripheral after the fw update.  The usual suspects are ALSA, pulseaudio.

guymcswain, 2021-02-15:
The raspberrypi forum has a lot of chatter about recent changes in the audio system:

> Since December 2020 Raspberry Pi with Desktop Operating Systems* have
utilised PulseAudio Sound Server which sits atop ALSA to handle ALL Audio I/O.

dgalland, 2021-02-15:
Thank you,
Yes I've seen changes in pulse audio in the last updates, but the debug is a bit beyond my skills !
I would like to take this opportunity to thank you for your work on pigpio.

guymcswain, 2021-03-11:
> I've seen changes in pulse audio in the last updates, but the debug is a bit beyond my skills !

Mine also.  You mention that this issue is non-blocking for you but have you since found a workaround or any additional insight into the problem?
Cheers.

dgalland, 2021-03-12:
I use the alternate source clock as you suggested to me.

guymcswain, 2021-03-19:
Thank you @dgalland  for your input.  

@RolfKausH , I'm closing this issue since there has been no dialogue with you.

closed 2021-03-19 by guymcswain