joan2937/pigpio #431: Should pigpio work on 64 bit Rasperry Pi OS?

timrowledge, 2021-02-15:
I recently had cause to try pigpio on a 64 bit OS system. It installed ok with `sudo apt-get install` and started happily. Basic setting of gpio pins seems to work; at least well enough to allow me to use NuScratch to drive  the plain gpio pin based parts of a PiMoRoNi Explorer HAT Pro. The i2c connections did not, however work at all.

I subsequently used the `pigs` command to make sure it wasn't my mistake and observed the same  thing; basic pin-waggling OK, i2C no. I double-checked the configuration of the interfaces just in case but that all appears to be correct.

guymcswain, 2021-02-16:
Please provide the i2c function you are calling and it's arguments.  In addition can you provide:
- OS revision
- output from `pigs hwver`
- /boot/config.txt

timrowledge, 2021-02-16:
Example routine I was trying to use - 
```
pigs prs 19 100
0
pigs prg 19
0
```

OS -
 ```
uname -a
Linux Pi-4-1-64 5.10.11-v8+ #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021 aarch64 GNU/Linux
```
pigpiod version 

```
pi@Pi-4-1-64:~ $ sudo pigpiod
pi@Pi-4-1-64:~ $ pigs hwver
0
```
/boot/config.txt - 
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=82

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
arm_64bit=1

```
Hope that helps ! Let me know if I can  find any more info or run any tests.

joan2937, 2021-02-16:
I can't see the pigpio version.

Try pigpiod -v

If it's not 78 you need to install the latest version and try your tests again.

guymcswain, 2021-02-16:
> pi@Pi-4-1-64:~ $ pigs hwver
0

This is a problem.  It implies there was no cpuinfo found and defaults to RPi model 1, I think.

guymcswain, 2021-02-16:
> # Enable audio (loads snd_bcm2835)
dtparam=audio=on

This might create a conflict with the PCM peripheral, used by default on pigpiod.

> dtparam=i2c_arm=on

You may want to experiment with this line commented.

guymcswain, 2021-02-16:
I'll be happy to investigate with my own hardware after the massive power outages are resolved in Texas.

timrowledge, 2021-02-16:
> On 2021-02-16, at 1:48 PM, joan2937 <notifications@github.com> wrote:
> 
> 
> I can't see the pigpio version.
> 
> Try pigpiod -v

Hmm, 71, so apparently out of date. It is, however, what you get from an apt-get install.

So, download the code zip, build, reboot.

Version is now 78, and 
pigs prs 19 100
250
pigs prg 19
100

So, excellent start. Restart NuScratch 64 bit... load Explorer HAT driver project...
Yup! It works. 

Thank you very much.

tim
PS Random sigline generator showing eerie AI tendencies ...
--
tim Rowledge; tim@rowledge.org; http://www.rowledge.org/tim
There are no stupid questions. But, there are a lot of inquisitive idiots.

timrowledge, 2021-02-16:
> On 2021-02-16, at 3:01 PM, Guy McSwain <notifications@github.com> wrote:
> 
> 
> I'll be happy to investigate with my own hardware after the massive power outages are resolved in Texas.

Good luck with that; not fun. We had an entire week without power and 3ft+ of snow a few years ago. Very unpleasant.

tim
--
tim Rowledge; tim@rowledge.org; http://www.rowledge.org/tim
Strange OpCodes: SDLI: Shift Disk Left Immediate

guymcswain, 2021-03-02:
Latest release, v79, has a fix for this:

> pi@Pi-4-1-64:~ $ pigs hwver
0

... and the ensuing confusion would have been avoided.  Thanks for your help.

closed 2021-03-02 by guymcswain