joan2937/pigpio #432: pigpio I2C requires root

eitch, 2021-02-22:
It seems that pigpio I2C communication only works with root permissions. Is there a way around this? I find this rather annoying, as i am using pigpio from a java application (pi4j) and thus the java application would require to run as root.

Is there a way to avoid this?

guymcswain, 2021-02-22:
You can access pigpio via a client interface - there are several Java clients listed on the pigpio home page - to avoid running your application as root.  You run pigpiod, as root, beforehand.  This can be set up as a service running in the background.

guymcswain, 2021-02-22:
Poking around the p4j-v2 repo I can see references to the pigpio socket interface (ie client).  However I also see a lot of references to, effectively, `pigpio.initialise()`.  To initialize the library you need root authorization.  Avoid initializing the library and instead access it via the daemon that is already running.

(I'm not a Java programmer so its difficult for me to fully understand the p4j stuff.)

eitch, 2021-03-03:
The linux i2c-tools don't need root privileges, so why should this library? That doesn't really make sense.

Of course we could use the daemon, but the thing about using the daemon, when coming from java is that this is another moving part. pi4j is also being used by students who are new to the entire programming ecosystem and the more moving parts, the more reasons for this not working will arise.

Then again, the pi4j v1 are directly using the /dev/i2c-1 file, so i will check to use that code.

closed 2021-03-03 by guymcswain