joan2937/pigpio #433: libgpiod support

pemensik, 2021-02-25:
I read interesting article about [GPIO remote in Fedora](https://fedoramagazine.org/raspberry-pi-zero-gpio-fedora/). I also experimented with RPi.GPIO on Fedora (made also [COPR package](https://copr.fedorainfracloud.org/coprs/pemensik/raspberry-pi/package/pigpio/)), but it just does not work with default kernel.

Is it possible to provide compatible pigpiod, which wouldn't use memory mapping, but would use libgpiod support enabled in vanilla kernels? Is its support planned or even in-progress?

I got recommendation on python [RPi.GPIO2](https://pypi.org/project/RPi.GPIO2/) library, which indeed works on Fedora kernels. But it would lack ability to map GPIO to remote host, like pigpio allows. Would it be possible to merge those two?

joan2937, 2021-02-26:
There is the lg archive which is a similar architecture to pigpio but is pure Linux (i.e. no use of special Raspberry Pi features).

See http://abyz.me.uk/lg/

github https://github.com/joan2937/lg

closed 2021-03-20 by guymcswain

doronbehar, 2023-04-20:
> There is the lg archive which is a similar architecture to pigpio but is pure Linux (i.e. no use of special Raspberry Pi features).

Hey, sorry to resurrect an old issue. It would have been nice if this was documented in both projects' web pages, perhaps with expanded explanations :). It's kind of hard to figure this out at the moment.

> By the way GPIO Remote in Fedora is actually using pigpio under the hood.

What is this software exactly? Is this gpiozero?