joan2937/pigpio #434: makedoc errors v79

guymcswain, 2021-03-02:
@joan2937 ,
As the last step in my release procedure for v79 I ran `makedoc` and got the following errors:

> *** making man pages ***
pigs
pigpiod
pig2vcd
pigpio.h
pigpiod_if.h
pigpiod_if2.h
char *addrStr used but not defined.
const used but not defined.
char *portStr used but not defined.
*addrStr defined but not used.
*portStr defined but not used.
char *addrStr used but not defined.
const used but not defined.
char *portStr used but not defined.
*addrStr defined but not used.
*portStr defined but not used.
*** making web pages ***
...

Maybe the header `pigpiod_if2` is missing a function declaration?  But there are no make errors in building the library and it runs the `x_pigpiod_if2` test without errors.

I've already pushed v79 to master.  I'll go ahead and push the newly built DOC files.  Please take a look at these for any errors.

joan2937, 2021-03-02:
Shall do, but as they don't appear to be that serious it may not be for a few days.  If this will cause a problem let me know and I'll give it more priority.

guymcswain, 2021-03-02:
No rush.  Thank you.

closed 2021-12-28 by guymcswain